qmk/qmk_firmware #16097: [Bug] Keychron Q2 default config has some issues on Linux

IvarWithoutBones, 2022-01-28:
## Describe the Bug

A few keys on my keychron Q2 (rev_0111) do not seem to be interpreted correctly with the stock config on Linux. 

Some keys output a lot of seemingly random keycodes, instead of the expected key. See this log from `sudo evtest` when pressing the right arrow key once:
```
^[^@Event: time 1643378313.448232, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e6
Event: time 1643378313.448232, type 1 (EV_KEY), code 100 (KEY_RIGHTALT), value 1
Event: time 1643378313.448232, -------------- SYN_REPORT ------------
Event: time 1643378313.472227, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70050
Event: time 1643378313.472227, type 1 (EV_KEY), code 105 (KEY_LEFT), value 1
Event: time 1643378313.472227, -------------- SYN_REPORT ------------
Event: time 1643378313.480230, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7004f
Event: time 1643378313.480230, type 1 (EV_KEY), code 106 (KEY_RIGHT), value 1
Event: time 1643378313.480230, -------------- SYN_REPORT ------------
Event: time 1643378313.496229, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e0
Event: time 1643378313.496229, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1643378313.496229, -------------- SYN_REPORT ------------
Event: time 1643378313.504226, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e3
Event: time 1643378313.504226, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1643378313.504226, -------------- SYN_REPORT ------------
Event: time 1643378313.512229, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e2
Event: time 1643378313.512229, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1643378313.512229, -------------- SYN_REPORT ------------
Event: time 1643378313.520226, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7002c
Event: time 1643378313.520226, type 1 (EV_KEY), code 57 (KEY_SPACE), value 0
Event: time 1643378313.520226, -------------- SYN_REPORT ------------
Event: time 1643378313.536230, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e6
Event: time 1643378313.536230, type 1 (EV_KEY), code 100 (KEY_RIGHTALT), value 0
Event: time 1643378313.536230, -------------- SYN_REPORT ------------
Event: time 1643378313.560490, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70050
Event: time 1643378313.560490, type 1 (EV_KEY), code 105 (KEY_LEFT), value 0
Event: time 1643378313.560490, -------------- SYN_REPORT ------------
Event: time 1643378313.568268, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7004f
Event: time 1643378313.568268, type 1 (EV_KEY), code 106 (KEY_RIGHT), value 0
```
The up arrow key outputs `ZXCVBNM<>?`, and sometimes selects some text in my terminal. The down and left keys work fine though.

Happens on both the mac and windows profiles. Note that i have only tested this on linux.

The rotary knob used to control volume perfectly as it should, but after rebasing my repo to commit 0f0e90971b6da2b1a125acc1620c3d33d0f1864c it seems to have regressed. My previous checkout was from the 26th, there the knob functioned as expected. All other issues were present there as well though.
```
Event: time 1643378475.594101, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70028
Event: time 1643378475.594101, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1643378475.594101, -------------- SYN_REPORT ------------
Event: time 1643378477.562313, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70035
Event: time 1643378477.562313, type 1 (EV_KEY), code 41 (KEY_GRAVE), value 1
Event: time 1643378477.562313, -------------- SYN_REPORT ------------
Event: time 1643378477.570073, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7001e
Event: time 1643378477.570073, type 1 (EV_KEY), code 2 (KEY_1), value 1
Event: time 1643378477.570073, -------------- SYN_REPORT ------------
Event: time 1643378477.578064, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7001f
Event: time 1643378477.578064, type 1 (EV_KEY), code 3 (KEY_2), value 1
Event: time 1643378477.578064, -------------- SYN_REPORT ------------
`12Event: time 1643378477.586067, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70020
Event: time 1643378477.586067, type 1 (EV_KEY), code 4 (KEY_3), value 1
Event: time 1643378477.586067, -------------- SYN_REPORT ------------
Event: time 1643378477.594064, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70021
Event: time 1643378477.594064, type 1 (EV_KEY), code 5 (KEY_4), value 1
Event: time 1643378477.594064, -------------- SYN_REPORT ------------
34Event: time 1643378477.602067, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70022
Event: time 1643378477.602067, type 1 (EV_KEY), code 6 (KEY_5), value 1
Event: time 1643378477.602067, -------------- SYN_REPORT ------------
Event: time 1643378477.610064, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70023
Event: time 1643378477.610064, type 1 (EV_KEY), code 7 (KEY_6), value 1
Event: time 1643378477.610064, -------------- SYN_REPORT ------------
56Event: time 1643378477.618067, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70024
Event: time 1643378477.618067, type 1 (EV_KEY), code 8 (KEY_7), value 1
Event: time 1643378477.618067, -------------- SYN_REPORT ------------
Event: time 1643378477.626065, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70025
Event: time 1643378477.626065, type 1 (EV_KEY), code 9 (KEY_8), value 1
Event: time 1643378477.626065, -------------- SYN_REPORT ------------
78Event: time 1643378477.634068, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70026
Event: time 1643378477.634068, type 1 (EV_KEY), code 10 (KEY_9), value 1
Event: time 1643378477.634068, -------------- SYN_REPORT ------------
Event: time 1643378477.642067, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70027
Event: time 1643378477.642067, type 1 (EV_KEY), code 11 (KEY_0), value 1
Event: time 1643378477.642067, -------------- SYN_REPORT ------------
Event: time 1643378477.650065, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7002d
Event: time 1643378477.650065, type 1 (EV_KEY), code 12 (KEY_MINUS), value 1
Event: time 1643378477.650065, -------------- SYN_REPORT ------------
90-Event: time 1643378477.658068, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7002e
Event: time 1643378477.658068, type 1 (EV_KEY), code 13 (KEY_EQUAL), value 1
Event: time 1643378477.658068, -------------- SYN_REPORT ------------
Event: time 1643378477.666064, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7002a
Event: time 1643378477.666064, type 1 (EV_KEY), code 14 (KEY_BACKSPACE), value 1
Event: time 1643378477.666064, -------------- SYN_REPORT ------------
Event: time 1643378477.682083, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70035
Event: time 1643378477.682083, type 1 (EV_KEY), code 41 (KEY_GRAVE), value 0
```
outputs the text: `1234567890-`. There's even more key events from a single press, but I think you get the point. Rotating the knob doesn't seem to do much either anymore. I haven't had the time to bisect this yet, but perhaps #16034 may have something to do with this? 
The two function keys below the knob also suffer from this issue where a lot of key events are send.

The only change I made locally was rebinding `KC_ESC` to `KC_GRAVE`, and `KC_CAPS` -> `KC_ESC`. here are the relevant lines:
```cpp
// Everything above and below this has been unchanged.
    [WIN_BASE] = LAYOUT_ansi_67(
        KC_GRV,  KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0,    KC_MINS,  KC_EQL,    KC_BSPC,          KC_MUTE,
        KC_TAB,  KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_LBRC,  KC_RBRC,   KC_BSLS,          KC_DEL,
        KC_ESC,  KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, KC_QUOT,             KC_ENT,           KC_HOME,
        KC_LSFT,          KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH,             KC_RSFT, KC_UP,
        KC_LCTL, KC_LGUI, KC_LALT,                            KC_SPC,                             KC_RALT, MO(_FN2), MO(_FN3),  KC_LEFT, KC_DOWN, KC_RGHT),
```
The mac layout didn't make a difference.

I also could not seem to get `KC_PRINT_SCREEN` bound to any key, it wouldn't even show up in `evtest`.

## System Information

 - Keyboard:  Keychron Q2
   - Revision :  rev_0111 (knob, ansi layout)
 - Operating system: NixOS Linux 5.15.16
 - AVR GCC version: 8.4.0

 - ARM GCC version: 10.2.1 20201103

 - QMK Firmware version: 0.15.18

 - No other keyboard related software installed

I am not extremely familair with QMK, apologies for the noise if this is because of an issue in my env. 

## Additional Context

IvarWithoutBones, 2022-01-28:
Reproduced this without any changes to the source:
```shell
> rm -rf ~/.config/qmk
> rm -rf ~/qmk_firmware
> qmk setup
> cd qmk_firmware
> nix-shell
> qmk config user.keyboard=keychron/q2/rev_0111
> qmk config user.keymap=ivarwithoutbones
> qmk compile -kb keychron/q2/rev_0111 -km ivarwithoutbones
> qmk flash -kb keychron/q2/rev_0111 -km ivarwithoutbones
```

sigprof, 2022-01-28:
The key event issue may be related to a ChibiOS bug in the STM32L432 chip support:

https://forum.chibios.org/viewtopic.php?f=35&t=6012

(apparently the rightmost matrix column on Keychron Q2 uses the `H3` pin, so it can be affected by that bug).

However, there might be something else at play here, because the current code for `keychron/q2/rev_0111` has `MCU = STM32L433`, and `platforms/chibios/boards/GENERIC_STM32_L433XC/configs/board.h` has `#define STM32L443xx`, which should be a suitable workaround for that ChibiOS bug.

IvarWithoutBones, 2022-01-28:
> The key event issue may be related to a ChibiOS bug in the STM32L432 chip support:
>
> https://forum.chibios.org/viewtopic.php?f=35&t=6012
>
> (apparently the rightmost matrix column on Keychron Q2 uses [...]

Thanks for the quick response! I'll see if I can try and see if I can hack anything together to make this work, although this seems a bit beyond my skill level 😅

Also, i can confirm https://github.com/qmk/qmk_firmware/pull/16034 caused the knob issue, after reverting [only this file](https://github.com/qmk/qmk_firmware/pull/16034/commits/62cd02d4c705ef5da3a663cb89ace2d3949d14a8#diff-75c40b72f2f0fdf24caa34deaa13e85d65cc56034106d0f8724bd0c13fe0806f) from https://github.com/qmk/qmk_firmware/pull/16034/commits/62cd02d4c705ef5da3a663cb89ace2d3949d14a8 it works perfectly again! Should i open a PR for that?

IvarWithoutBones, 2022-01-28:
A few more things: when i quickly spam the right (broken) column, the entire keyboard seems to crash. No input will be accepted until I reboot it, and the RGB hangs. Is there some kind of debug log / stacktrace for QMK? Might help identifying this issue.

I can reproduce this on master as well.

noroadsleft, 2022-01-29:
> Also, i can confirm `#16034` caused the knob issue, after reverting [only this file](https://github.com/qmk/qmk_firmware/pull/16034/commits/62cd02d4c705ef5da3a663cb89ace2d3949d14a8#diff-75c40b72f2f0fdf24caa34deaa13e85d65cc56034106d0f8724bd0c13fe0806f) from [`62cd02d`](https://github.com/qmk/qmk_firmware/pull/16034/commits/62cd02d4c705ef5da3a663cb89ace2d3949d14a8) it works again! Should i open a PR for that? There's still a few issues, for example when pressing the knob button mutes your volume as expected, but it still inputs some random numbers. Still better than nothing though :)

The issue that commit fixes is that the codeblock being edited was not properly formatted when submitted to QMK. If you compile from a branch that includes 16034, you need to move the keycodes intended for the encoder rotation action to either side of the keycode assigned to the encoder click (e.g. Volume Down, then Mute, then Volume Up).

Sadly, the rest of your issues strike me as some sort of hardware problem (either component failure, or some bug in a dependency somewhere).

sigprof, 2022-01-29:
> Also, i can confirm #16034 caused the knob issue, after reverting [only this [...]

The `LAYOUT_all` macro that is modified by that change is used only in the `via` keymap.  Is your personal keymap based on that?  In general, this is something that should not be done, because the VIA-related code stores the actual keymap in the EEPROM, and sometimes reflashing the updated firmware does not reset the EEPROM keymap, therefore you may be actually testing some old keymap instead of what you flashed.  And for this particular board the `via` keymap contains some special code that remaps the encoder actions to some fake keymap locations.

Does the encoder rotation still misbehave if you try the `default` keymap with the current QMK code?  (By default it should control the system volume.)

As for your problems with the right column, I unfortunately don't have any hardware with the STM32L432 chip, but I tested the `keychron/q2/rev_0111` firmware on a development board with the STM32L433CCT6 chip, and the problematic `H3` pin seems to work there — but the twist is that the `H3` pin is also `BOOT0`, therefore it is somehow connected to the “reset to bootloader” button under the spacebar on your board, and there are slightly more possibilities of things going wrong with that than for any other column pin. So I would suggest this:
1. Take out all rightmost switches that misbehave (two keys under the knob, and the cursor up and right keys); check them for bent pins, and try pressing the knob without those switches inserted — sometimes a pin bent in a specific way can cause similar issues.
2. Flash the stock firmware from https://www.keychron.com/blogs/archived/how-to-factory-reset-or-flash-your-qmk-via-enabled-keychron-q2-keyboard; for Linux you will need to do run `dfu-util` manually, using a command like this after connecting your board in the bootloader mode:
```sh
   dfu-util -d 0483:df11 -a 0 -s 0x08000000:leave -D ~/Downloads/Keychron-Q2-ANSI-Knob-Version-Firmware_V1.0.2.bin
```
3. If you still get the same “whole row gets activated” issue on the stock firmware without any switches inserted to problematic positions, probably some hardware repair or replacement is needed.

selmanj, 2022-01-31:
Just wanted to chime in that I'm hitting the same issue as @IvarWithoutBones where the arrow keys are inserting garbage (which makes it less likely to be a hardware failure). The stock firmware has no such issue.

charlesnchr, 2022-02-01:
I am experiencing this issue too. I have tried using the keyboard, compiling and flashing the firmware on Window/Linux/macOS, and this bug occurs on each platform. Stock firmware does not have this issue.

BenVosper, 2022-02-03:
I'm also having this issue on the Rev0112 (ISO, no-knob). The default keymap on `qmk_firmware` master gets me the broken rightmost row as described above. Flashing back to the factory firmware as described by @sigprof [above](https://www.keychron.com/blogs/archived/how-to-factory-reset-or-flash-your-qmk-via-enabled-keychron-q2-keyboard) fixed the issue.

Does anyone know what the issue might be?

strange-moth, 2022-02-03:
I can report this behavior as well on rev_0110 (ansi, no knob)
No problems on stock firmware, built and flashed the default qmk keymap on Linux, the same garbled key output appeared when using the keyboard with both Linux and MacOS, and finally after reflashing the stock firmware from keychron's website the issue vanishes

IvarWithoutBones, 2022-02-03:
Thanks for the help everyone, I really appreciate it.

> The LAYOUT_all macro that is modified by that change is used only in the via keymap. Is your personal keymap based on that? In general, this is something that should not be done, because the [...]

Oops, I shouldn't have used that as an example. I only tried that macro when debugging to see if it magically fixed everything. I've never used via or its keymap, and ill avoid it in the future. Thanks for the advice. I actively use the default keymap, where all issues occur exactly the same.

> therefore it is somehow connected to the “reset to bootloader” button under the spacebar on your board, and there are slightly more possibilities of things going wrong with that than for any other column pin.

Aha, I never considered that before. After checking all my switches though, they all seemed fine. I tried putting in a few switches I've never used before on the problematic keys, but that didn't seem to change anything.

I tried the stock firmware as suggested, and there everything works as expected. This definitly seems like a qmk issue.

> The issue that commit fixes is that the codeblock being edited was not properly formatted when submitted to QMK. If you compile from a branch that includes 16034, you need to move the keycodes [...]

That makes sense, Thanks! This issue occurs with the default generated keymap file though, any chance we could get a fix in master?

selmanj, 2022-02-04:
I believe PR #16200 fixes the issues with the arrow keys activating the entire lower row - all it does is re-add `matrix.c` as a SRC for the Q2.

selmanj, 2022-02-04:
Also, I only added the file to rev_0111 (ansi, knob) as that's the only board I can test with. It'd be good if someone could confirm it works with the other revisions as well.

BenVosper, 2022-02-04:
@selmanj I can confirm that your change in #16200 fixes the issue on my Rev 0112 (ISO, no-knob). The arrow keys (as well as the whole rightmost column) now work as expected. Thanks very much!

lalalademaxiya1, 2022-02-07:
To be honest, at the beginning of release, we used our own matrix scanning code (matirx. C) in the master branch, maybe someone had optimized the code and deleted "SRC += matrix.c" in the rule.mk, so the firmware compiled with the master branch will produce this bug #16097. You can get the code from [,](https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/keychron/q2) or replace R7 from 10K to 100k, then the PH3 (boot0) can be used as a general I/O and we don't need change the current code , Thank you.
[image]

lalalademaxiya1, 2022-02-07:
To be honest, at the beginning of release, we used our own matrix scanning code (matirx. C) in the master branch, maybe someone had optimized the code and deleted "SRC += matrix.c" in the rule.mk, so the firmware compiled with the master branch will produce this bug https://github.com/qmk/qmk_firmware/issues/16097. You can get the code from [q2-keychron](https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/keychron/q2) or replace R7 from 10K to 100k, then the PH3 (boot0) can be used as a general I/O and we don't need change the current code , Thank you~
Like that:
[image]

tzarc, 2022-02-08:
Can I get you folks to try applying https://github.com/qmk/qmk_firmware/commit/aebece55dd925528f8958a50b4b138b92de27a1e instead of #16200? Trying to sort out a more maintainable solution.

selmanj, 2022-02-09:
@tzarc I can confirm that commit fixes the issue on my rev_0111 Q2.

closed 2022-02-09 by tzarc

IvarWithoutBones, 2022-02-10:
Thanks for the fix, this works perfectly!

The knob issue also seems to stem from my environment, as it works perfectly fine on the stable branch of my distro.