qmk/qmk_firmware #16258: [QUESTION] What C mode style is used for editing keymaps

mcp292, 2022-02-07:
I am using (spac)emacs and would like to be able to set the `c-mode` style with `c-set-style`. 

Is there actually a supported style used or is it just convention to make everything fit on the screen?

Is this documented anywhere?

tzarc, 2022-02-07:
You'll probably be best off inferring from:
https://github.com/qmk/qmk_firmware/blob/master/.editorconfig
https://github.com/qmk/qmk_firmware/blob/master/.clang-format

mcp292, 2022-02-20:
Thank you for the links.

I settled on using `text-mode` as a workaround. This prevents most of the re-alignment that happens when editing keycodes in `c-mode`.

mcp292, 2022-02-20:
I will close this issue for now, but if anyone has better suggestions please add them here.

closed 2022-02-20 by mcp292