qmk/qmk_firmware #16272: 编码器不能设置修饰键+基础键的组合键。The encoder cannot set the combination key of modifier key + base key

jeromeli002, 2022-02-08:
编码器不能设置修饰键+基础键，但是修饰键没问题，组合起来就只能触发基础键，比如 ctrl+E 只能触发E而不是ctrl+E。编码器是映射的矩阵位置，不是写在via/keymap.c里面的

The encoder cannot set modifier keys + basic keys, but there is no problem with modifier keys, they can only trigger basic keys when combined, for example, ctrl+E can only trigger E instead of ctrl+E. The encoder is the mapped matrix position, not written in via/keymap.c

jeromeli002, 2022-02-10:
> 你说的编码器是指什么？

ec11编码器 固件是映射的在矩阵里面的。可以方便via修改，发现设置单个按键是漆作用的，但是组合键就不行了，只能触发一个键，不能实现组合键的功能

givedrug, 2022-02-10:
> > 你说的编码器是指什么？
> 
> ec11编码器 固件是映射的在矩阵里面的。可以方便via修改，发现设置单个按键是漆作用的，但是组合键就不行了，只能触发一个键，不能实现组合键的功能

你当前的编码器配置是写在哪里的？能贴出来看下吗？

jeromeli002, 2022-02-10:
> > > 你说的编码器是指什么？
> > 
> > 
> > ec11编码器 固件是映射的在矩阵里面的。可以方便via修改，发现设置单个按键是漆作用的，但是组合键就不行了，只能触发一个键，不能实现组合键的功能
> 
> 你当前的编码器配置是写在哪里的？能贴出来看下吗？

keymap.c这个文件里面的

[image]

jeromeli002, 2022-02-10:
> > > > > 你说的编码器是指什么？
> > > > 
> > > > 
> > > > ec11编码器 固件是映射的在矩阵里面的。可以方便via修改，发现设置单个按键是漆作用的，但是组合键就不行了，只能触发一个键，不能实现组合键的功能
> > > 
> > > 
> > > 你当前的编码器配置是写在哪里的？能贴出来看下吗？
> > 
> > 
> > keymap.c这个文件里面的
> > [image]
> 
> 可以试试这样：
> 
> ```
> if (clockwise) {
>         // 顺时针
>         register_code(KC_LEFT_SHIFT);
>         register_code(KC_LEFT_CTRL);
>         register_code(KC_LEFT_ALT);
>         tap_code(KC_K);
>         unregister_code(KC_LEFT_SHIFT);
>         unregister_code(KC_LEFT_CTRL);
>         unregister_code(KC_LEFT_ALT);
>     } else {
>         //逆时针
>     }
> ```

这样就不能通过via自定义旋钮了啊

skyjun, 2022-02-15:
`bool encoder_update_user(bool clockwise)
{
    if (clockwise) {
        tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 5));
    } else {
        tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 6));
    }

    return true;
}`

jeromeli002, 2022-02-28:
> ` bool encoder_update_user(bool clockwise) { if (clockwise) { tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 5)); } else { tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 6)); }
> 
> ```
> return true;
> ```
> 
> } `

多旋钮怎么加呢？这才一个旋钮吧

7-liang, 2022-02-28:
> > ` bool encoder_update_user(bool clockwise) { if (clockwise) { tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 5)); } else { tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 6)); }
> > ```
> > return true;
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > } `
> 
> 多旋钮怎么加呢？这才一个旋钮吧

我把 index 去掉了，加一个uint8_t index参数

jeromeli002, 2022-02-28:
> 我把 index 去掉了，加一个uint8_t index参数

bool encoder_update_user(uint8_t index, bool clockwise) {
if (index == 0) {/* 编码器1 */
if (clockwise) {
tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 5));
} else {
tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 6));
}
} else if (index == 1) { /* 编码器2 */
if (clockwise) {
tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 1, 5));
} else {
tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 1, 6));
}
}
return true;
}
这样吗？大佬

7-liang, 2022-02-28:
> bool encoder_update_user(uint8_t index, bool clockwise) { if (index == 0) {/* 编码器1 _/ if (clockwise) { tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 0, 5)); } else { [...]

不是大佬，白菜一颗，是这样，看看是否满足你的需求

0,5   0,6   0是row   5,6是col ，在keymap里找空位加上两个，相应row col 改一下

jeromeli002, 2022-02-28:
> 不是大佬，白菜一颗，是这样，看看是否满足你的需求
>
> 0,5 0,6 0是row 5,6是col ，在keymap里找空位加上两个，相应row col 改一下

矩阵位置我知道，上面哪有复制一份是这样吗？这样旋钮可以使用组合键吗？原来那样旋钮不能使用组合键，设置组合键也只能触发一个键 修饰键不能触发。

7-liang, 2022-02-28:
> 

配合切层使用是可以的，我的编码器第一层是鼠标滚轮，第二层音量大小，第三层home end，如果你需要配合ctrl alt什么的，试试看，应该行吧，我没试过，话说编码器需要配合ctrl alt什么的用么

jeromeli002, 2022-02-28:
> > 
> 
> 配合切层使用是可以的，我的编码器第一层是鼠标滚轮，第二层音量大小，第三层home end，如果你需要配合ctrl alt什么的，试试看，应该行吧，我没试过，话说编码器需要配合ctrl alt什么的用么

单独按键是没问题的，就是配合修饰键就不行了，其他按键是可以配合修饰键，唯独旋钮映射的矩阵不行
一些软件快捷键需要配合修饰键用啊

7-liang, 2022-02-28:
> 抄　送：李钧 ***@***.***>; Comment ***@***.***>
> 主　题：Re: [qmk/qmk_firmware] 编码器不能设置修饰键+基础键的组合键。The encoder cannot set the combination key of modifier key + base key (Issue #16272)

应该是可以的，上机试一下吧

--------------------------------------------------------------------------
 顺 祝 商 祺
--------------------------------------------------------------------------
李  钧

------------------------------------------------------------------

drashna, 2022-03-10:
I appoligize for buttin in, since I don't speak the language, but the issue here is that `tap_code` only handles basic keycodes.  It does not handled shifted keycodes, nor modded keycodes, nor any "special"/quantum keycodes. 

tap_code16 handles modded keycodes, but no special keycodes. 

There are a couple of ways to properly support this behavior, but, tap_code is not the way. Nor is dynamic_keymap_get_keycode. 

That said, the develop branch has added support for encoder maps (eg, keymaps but for encoders) (see a number of my keymaps for details). 

Additionally, a 'good' way to implement support for this right now, that works with VIA can be seen here, with the "encoder actions":
https://github.com/qmk/qmk_firmware/tree/master/keyboards/atlantis/ak81_ve/keymaps/via

givedrug, 2022-03-29:
我试了一下，可以直接在VIA中使用宏，来实现你的要求
首先导入json（keymap.c中可以随便实现一个按键，反正后面都会被via的json覆盖），并创建宏
{KC_LSFT,KC_LCTL,KC_LALT,KC_K}
[image]
然后save之后load进来就可以了，通过按键监控，可以看到，逆时针的时候已经可以同时触发四个按键
[image]
如果需要更改，随时修改宏配置即可

closed 2022-07-28 by github-actions[bot]