qmk/qmk_firmware #16273: [Bug] Introduced in 0.14.0: split kb (elite-c, ergodash) other half not working

freeo, 2022-02-08:
After git pulling the latest qmk my right half wasn't working anymore at all (left is master). I've traced this to version 0.14.0. Latest working version is 0.13.38 with the same json layout file. Compile & flash works fine - `qmk clean -a` was used everytime.

## System Information

 - Keyboard:  
   - Revision (if applicable): ergodash/rev1  
 - Operating system: PopOS 21.10 (Ubuntu)
 - AVR GCC version: 5.4.0

 - ARM GCC version: (15:9-2019-q4-0ubuntu2) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 - QMK Firmware version: 0.14.0 (see above, 0.13.38 works)

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:
   - [x] AutoKey

## Additional Context

A bit unusual controller setup:

Left half (master): Elite-C
Right half: Pro Micro

Erovia, 2022-02-08:
Have you reflashed both sides with the same firmware?
For keymap updates only the master side needs flashing, but for version updates it's recommended to re-flash both sides.

vozhyk-, 2023-05-29:
I've experienced the same with `ergodash/rev1` when upgrading from 0.12.37 to 0.18.4. Flashing the right half fixed the problem.