qmk/qmk_firmware #16346: Move Keyboards & Keymaps into submodule

ThePreviousOne, 2022-02-13:
I'm sure this has been brought up before but I can't find the issue.
Keyboards and keymaps are not part of the firmware. Personally I use the commit history to determine if I need to update my local branch or not, but the majority of commits are either new keyboards & keymaps or documentation changes. I won't advocate moving documentation as thats really just extra work with little benefit considering that documentation changes can be important when there is a new feature added, but the same is not true of keyboards or keymaps.

The commit history is the closest thing we have to a change log, which is important in a project this big to keep track of new features.
I realize maintaining an actual change log is extra work and I wouldn't ask anyone to do it, never the less it is important to have one in some form.

filterpaper, 2022-02-14:
There are changes proposed in #11269 to support out-of-tree userspace.

drashna, 2022-03-10:
This is a topic that comes up frequently enough.  

The biggest issue here isn't just moving stuff, it's also infrastructure stuff.  There is a lot that needs to be done to be able to move the keyboards and keymaps, without breaking a bunch of stuff.

ThePreviousOne, 2022-03-10:
In what way, I cant see how it would be anymore complex than adding a line to the Docs about cloning the submodule 'if desired'.
Sorry if that sounds argumentative, I'm really just curious and trying to understand.

closed 2022-07-16 by github-actions[bot]