qmk/qmk_firmware #16365: STM32F401很多引脚不能作为矩阵使用吗？加到矩阵中不能触发按键。Can't many pins of STM32F401 be used as a matrix? Adding to the matrix does not trigger keys.

jeromeli002, 2022-02-16:
STM32F401很多引脚不能作为矩阵使用吗？加到矩阵中不能触发按键。如图所示这样连接后部分按键不能使用

Can't many pins of STM32F401 be used as a matrix? Adding to the matrix does not trigger keys. Some buttons cannot be used after connecting as shown in the picture

[image]

rgoulter, 2022-02-16:
A10, B6, B7 should be fine. I'm using them in a keyboard which reads these pins for input. https://github.com/rgoulter/keyboard-labs/releases/tag/x1-rev2021.5 and qmk firmware under https://github.com/rgoulter/qmk_firmware/tree/rgoulter-keyboards-and-layouts/keyboards/rgoulter/minif4_36

With B6, B7, these are by-default used for I2C. So I had code snippet in a `config.h` like:

```
#ifdef OLED_DRIVER_ENABLE
/* B8, B9 instead of B6, B7 */
#define I2C1_SCL 8
#define I2C1_SDA 9
#endif
```.

jeromeli002, 2022-02-16:
> A10, B6, B7 should be fine. I'm using them in a keyboard which reads these pins for input. https://github.com/rgoulter/keyboard-labs/releases/tag/x1-rev2021.5 and qmk firmware under https://github.com/rgoulter/qmk_firmware/tree/rgoulter-keyboards-and-layouts/keyboards/rgoulter/minif4_36
> 
> With B6, B7, these are by-default used for I2C. So when using I2C, I had code snippet in a `config.h` like:
> 
> ```
> #ifdef OLED_DRIVER_ENABLE
> /* B8, B9 instead of B6, B7 */
> #define I2C1_SCL 8
> #define I2C1_SDA 9
> #endif
> ```
> 
> (but with QMK's board definition, I2C isn't enabled by default https://github.com/qmk/qmk_firmware/blob/master/platforms/chibios/boards/BLACKPILL_STM32_F401/configs/mcuconf.h#L124).
这样的ROW2（A10）开始后面的就不能触发了，col都正常的，我直接用STM32F401开发板
Such ROW2 (A10) cannot be triggered at the beginning, col is normal, I directly use the STM32F401 development board

#define MATRIX_COL_PINS {B14, B15, A8, B8, B9}
#define MATRIX_ROW_PINS {A15, B3, A10, B6, B7}

/* COL2ROW, ROW2COL*/
#define DIODE_DIRECTION COL2ROW

#define MATRIX_IO_DELAY 5

drashna, 2022-03-10:
IIRC< there is an issue with B6 and i2c on the f401.  

as for A!0, if this is a blackpill, you may need a pullup resistor on it, to get it to load the bootloader correctly.

closed 2022-07-17 by github-actions[bot]