qmk/qmk_firmware #16384: [Bug] Keyboard Unresponsive After Flash. Reboot Required.

JonatannossreP, 2022-02-16:
Seemingly at random the keyboard stops responding to any input after flashing it with a working firmware. Not every time you flash, but sometimes. Keyboard is unusable until my computer is restarted. 

This all started with this message on Discord: I have handwired keyboard with a all custom hardware and software layout that uses a at90usb1286 MCU. If I get everything to compile without any errors and the device shows up on my computer and installs correctly, yet nothing works. What is the most likely issue and where should I start checking where something might have went wrong?
I had the whole keyboard working before, but because of several circumstances I do not have the the old .hex file that did work, and my old files did not work with the new qmk version, I had not updated my keyboard firmware since before QMK MSYS was a thing, so I know it is not a hardware issue.

After successfully remaking a basic keyboard and getting it work, little by little I copy over more and more from my old files over to the confirmed working ones. Then it stopped working, but I was quite sure I had not really changed anything important, I even minuscule changes to rule out all issues. So I just used "undo", compiled it again with the files that previously worked, and it still did not work! I restarting my computer because there must be some PC issue that might temporarily fix itself if I do. And what do you know, the keyboard worker again! So it turns out that the keyboard seemingly at random stops working after flashing. This never happened in previous versions of QMK for me, and now it happens again and again after testing to flash, it all seems quite random to me. Strange. I even had the "NO_USB_STARTUP_CHECK = yes" that was said to fix a similar issue years ago: [](https://www.reddit.com/r/olkb/comments/lc0nq8/keyboards_randomly_unresponsive_until_os_reboot/). I have had that tag set for longer than and had completely forget about it, but now it does not seems to help however. I also tried removing it in case it was outdated by now and might have cause some issue, but to no help.

## System Information
 - Operating system: Windows 10
 - AVR GCC version: 8.4.0
 - ARM GCC version: 10.1.0		
 - QMK Firmware version: Command does not work for me, but updated today
 - Any keyboard related software installed?: No, closest I have to affecting input is rawaccel for my mouse and custom global hotkeys for AIMP

drashna, 2022-03-10:
Could you post a link to your config, as well as the output of `qmk doctor`?

JonatannossreP, 2022-03-11:
> Could you post a link to your config, as well as the output of `qmk doctor`?
https://www.dropbox.com/s/94n3dxkvvgkfqhw/zyrhen_joppy_2022-03-11.zip?dl=0
https://www.dropbox.com/s/thwni40zih6fl44/qmk-doctor_joppy_2022-03-11.jpg?dl=0