qmk/qmk_firmware #16391: [Bug] (Debug logs at end) Problem with combo-mod (eg combo that resolves to KC_LSFT) into mod-tap (eg LCTL_T(KC_P)): does not capitalize.

sdgu, 2022-02-17:
## Describe the Bug
(see also debug info below)
CTL_P: LCTL_T(KC_P)
CMB_SFT: combo of "." and "o" that resolves to KC_LSFT
[Keymap here.](https://github.com/sdgu/qmk_firmware/tree/sdgukeymaps/keyboards/boardwalk/keymaps/sdgu)

1) Trying to type capital "P" gives me trouble. Done quickly: "CMB_SFT down, CTL_P down, CMB_SFT up, CTL_P up" gives me lowercase "p", but I want capital "P".
2) Using my actual shift key at a similar speed doesn't have this issue; "KC_LSFT down, CTL_P down, KC_LSFT up, CTL_P up" gives me capital "P".
3) Also, if I try to capitalize a non-mod-tap letter, say "U", with my combo, I don't have this problem. That is, "CMB_SFT down, KC_U down, CMB_SFT up, KC_U up", gives me capital "U".

[Video example (might be a bit loud).](https://www.youtube.com/watch?v=HrNvVsExouo)

Based on what I understand from the debugging, I think a solution could be to keep a timer of how long ago my combo-shift was de-pressed, and if the tap of the mod-tap is triggered while the timer is under a certain number, send the shifted version. But I'm wondering if what I'm experiencing is due to a "bug" or if it's an implementation limitation.

## System Information

 - Keyboard:  Boardwalk, Technik_o
   - Revision (if applicable):  
 - Operating system: Windows, MacOS
 - AVR GCC version: 8.4.0
 - ARM GCC version: 10.1.0
 - QMK Firmware version: 0.15.21
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
Some debugging. I annotated some stuff with my understanding of what's happening. 

The following gives lowercase "p", but I want capital "P".
```
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0010000000000000
    03: 0000000000000000
    04: 0000000000000000
    
  > r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0001000000000000
    02: 0010000000000000
    03: 0000000000000000
    04: 0000000000000000
  > keyboard_report: 06 02 00 00 00 00 00 00 

    // matrix says: (02, 2) and (01, 3) are down
    // keyboard says: KC_LSFT 
    
    r/c 0123456789ABCDEF
  > 00: 0000000000000000
    01: 0001100000000000
    02: 0010000000000000
    03: 0000000000000000
    04: 0000000000000000
    
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0001100000000000
    02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000100000000000
    02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    keyboard_report: 06 00 00 00 00 00 00 00 

    // (01, 4), corresponding to LCTL_T(KC_P), is down according to the matrix
    // matrix says: (01, 4) is down
    // keyboard says:
    
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
    02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
  > keyboard_report: 06 00 00 00 08 00 00 00 // matrix says:                          // keyboard says: KC_P
    keyboard_report: 06 00 00 00 00 00 00 00 

    // matrix says: 
    // keyboard says: 
```

The following gives capital "U", as expected.
```
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0010000000000000
    03: 0000000000000000
    04: 0000000000000000
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0001000000000000
  > 02: 0010000000000000
    03: 0000000000000000
    04: 0000000000000000
    keyboard_report: 06 02 00 00 00 00 00 00 

    // combo for shift 
    // matrix says: (02, 2) and (01, 3) are down
    // keyboard says: KC_LSFT
    
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0001000000000000
    02: 0010100000000000
    03: 0000000000000000
    04: 0000000000000000
    keyboard_report: 06 02 00 00 00 01 00 00 

    // (02, 4), corresponding to "U", which is a standard keycode, is hit
    // matrix says: (02, 2), (03, 3), (02, 4) are down
    // keyboard says: KC_LSFT + KC_U
    
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0001000000000000
    02: 0000100000000000
    03: 0000000000000000
    04: 0000000000000000
    
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
    02: 0000100000000000
    03: 0000000000000000
    04: 0000000000000000
  > keyboard_report: 06 00 00 00 00 01 00 00 

    // matrix says: (02, 4) is down
    // keyboard says: KC_U
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    keyboard_report: 06 00 00 00 00 00 00 00 

    // matrix says: 
    // keyboard says: 
```

Using actual shift instead of a combo gives expected results:

Capital "P", rolling key sequence
```
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 1000000000000000
    04: 0000000000000000
    keyboard_report: 06 02 00 00 00 00 00 00 
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000100000000000
  > 02: 0000000000000000
    03: 1000000000000000
    04: 0000000000000000
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000100000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    keyboard_report: 06 02 00 00 08 00 00 00 
    keyboard_report: 06 00 00 00 08 00 00 00 
    keyboard_report: 06 00 00 00 00 00 00 00 
```

Capital "U", rolling key sequence
```
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 1000000000000000
    04: 0000000000000000
    keyboard_report: 06 02 00 00 00 00 00 00 
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000100000000000
    03: 1000000000000000
    04: 0000000000000000
    keyboard_report: 06 02 00 00 00 01 00 00 
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000100000000000
    03: 0000000000000000
    04: 0000000000000000
    keyboard_report: 06 00 00 00 00 01 00 00 
    
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
    02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
  > keyboard_report: 06 00 00 00 00 00 00 00 
```

jbernie2, 2022-10-14:
I have also had this exact problem.
I am using a combo of `KC_D` and `KC_F` as a shift key, and when I try to use it in combination with mod tap keys, it takes significantly longer to work. It still works, I just have to hold the combo down for a much longer time.

source files: https://github.com/jbernie2/dotfiles/tree/combo-mods/keyboards/ergodox_ez/src

I am happy to provide more details if needed.

sigprof, 2022-10-14:
Probably this piece of code does not work correctly when a modifier keycode comes from a combo:
https://github.com/qmk/qmk_firmware/blob/06a7a1c205f43c0c4c435d48de5ab3893398b19d/quantum/action_tapping.c#L228-L250

The intent of that code is to process key release events immediately (to avoid accidentally triggering autorepeat while waiting for the tapping term to expire), *except release event for modifier keys* — the second part uses `layer_switch_get_action(event.key)` to determine the key action, but events injected by combos have some dummy information in `event.key` (they use the extra `keycode` field instead), therefore the test does not detect the combo result as a modifier key, and its release comes through immediately before the hold/tap detection completes.

jbernie2, 2022-10-28:
I didn't fix the bug, but I did find a work around that is working for me.
In addition to using a combo for shift, I also added combos for the shift combo + any mod tap keys

I used the combo extension described here: http://combos.gboards.ca/docs/combos/

so now my shift combos look like this
```
COMB(df_shift,     KC_LSFT, MT_C(KC_D), MT_S(KC_F))
COMB(jk_shift,     KC_LSFT, MT_S(KC_J), MT_C(KC_K))

SUBS(caps_a_1,     "A", MT_C(KC_K), MT_S(KC_J), MT_G(KC_A))
SUBS(caps_a_2,     "A", MT_S(KC_J), MT_C(KC_K), MT_G(KC_A))
SUBS(caps_s_1,     "S", MT_C(KC_K), MT_S(KC_J), MT_A(KC_S))
SUBS(caps_s_2,     "S", MT_S(KC_J), MT_C(KC_K), MT_A(KC_S))
SUBS(caps_d_1,     "D", MT_C(KC_K), MT_S(KC_J), MT_C(KC_D))
SUBS(caps_d_2,     "D", MT_S(KC_J), MT_C(KC_K), MT_C(KC_D))
SUBS(caps_f_1,     "F", MT_C(KC_K), MT_S(KC_J), MT_S(KC_F))
SUBS(caps_f_2,     "F", MT_S(KC_J), MT_C(KC_K), MT_S(KC_F))
SUBS(caps_g_1,     "G", MT_C(KC_K), MT_S(KC_J), MT_H(KC_G))
SUBS(caps_g_2,     "G", MT_S(KC_J), MT_C(KC_K), MT_H(KC_G))
SUBS(caps_h_1,     "H", MT_S(KC_F), MT_C(KC_D), MT_H(KC_H))
SUBS(caps_h_2,     "H", MT_C(KC_D), MT_S(KC_F), MT_H(KC_H))
SUBS(caps_j_1,     "J", MT_S(KC_F), MT_C(KC_D), MT_S(KC_J))
SUBS(caps_j_2,     "J", MT_C(KC_D), MT_S(KC_F), MT_S(KC_J))
SUBS(caps_k_1,     "K", MT_S(KC_F), MT_C(KC_D), MT_C(KC_K))
SUBS(caps_k_2,     "K", MT_C(KC_D), MT_S(KC_F), MT_C(KC_K))
SUBS(caps_l_1,     "L", MT_S(KC_F), MT_C(KC_D), MT_A(KC_L))
SUBS(caps_l_2,     "L", MT_C(KC_D), MT_S(KC_F), MT_A(KC_L))
SUBS(caps_quote_1, "\"", MT_S(KC_F), MT_C(KC_D), MT_G(KC_QUOTE))
SUBS(caps_quote_2, "\"", MT_C(KC_D), MT_S(KC_F), MT_G(KC_QUOTE))
```

Each key has two combos associated with it so that I can press the shift combo in either order, but have to press the alpha key I am shifting last.

source: https://github.com/jbernie2/dotfiles/blob/combo-shift/keyboards/ergodox_ez/src/combos.def