qmk/qmk_firmware #16450: Help with flashing via the command line

kip93, 2022-02-26:
## Description

I am currently trying to flash my corne keyboard using the command line in Linux. I've tried installing and using QMK in many different ways, with no luck.
 * Used `nix` and the provided `shell.nix`
 * Installed it via `pip`
 * Tried the `pacman` version
 * Built it from source
 * Tested several older versions as well

No matter what I do, the flashing just does not detect the keyboard. Digging deeper, it seems that the issue is the system itself not detecting the keyboard after pressing the reset button, so QMK has nothing to work with. Yet, I cannot find anything online that worked.
 * Added the user to the `dialout` and `plugdev` groups
 * Found that there is a weird bug where `ModemManager` causes conflicts, so I got rid of it to no avail
 * Some people suggest to use `avrdude` directly, but since the board itself is not being detected it did not help either
 * Obviously I also read https://docs.qmk.fm/#/newbs_flashing?id=flash-your-keyboard-from-the-command-line & https://docs.qmk.fm/#/flashing?id=caterina but I found no relevant info

At this point, I am a complete loss, so any help would be greatly appreciated!

## Example

```sh
git clone --recurse-submodules -- 'https://github.com/qmk/qmk_firmware.git'
cd './qmk_firmware/'
mkdir -p '/etc/udev/rules.d/'
sudo cp -rf './util/udev/'* '/etc/udev/rules.d/'
sudo find /etc/udev/rules.d/ /usr/lib/udev/rules.d/ -type f -exec sudo chmod 0644 '{}' ';'
sudo reboot
nix-shell --run 'qmk compile -kb crkbd -km default'
nix-shell --run 'qmk flash -kb crkbd -km default'
```

## System information

 - Keyboard: crkbd
   - Revision (if applicable): rev1
 - Operating system: Manjaro Linux
 - AVR GCC version: 8.4.0
 - ARM GCC version: 10.2.1 20201103
 - QMK Firmware version: 0.16.0

## Additional context

I've tried it on a Windows machine and QMK Toolbox and I can flash the keyboard no problem, so it is not a hardware issue.

sigprof, 2022-02-26:
What kind of controllers do you have on your crkbd? If you are using Elite-C (or some other similar controller board that uses the `atmel-dfu` bootloader), you need to add the following line to the `rules.mk` file in your custom keymap:
```make
BOOTLOADER = atmel-dfu
```
By default `crkbd` assumes that you are using a Pro Micro board with the Caterina bootloader (`BOOTLOADER = caterina`), therefore `qmk flash` expects to see that bootloader; unlike QMK Toolbox, the CLI does not perform autodetection of the bootloader type.

Specifying the correct bootloader type is also required to make features like Bootmagic Lite or the `RESET` keycode to work properly.

You can also add the `-e BOOTLOADER=atmel-dfu` option to the `qmk compile` or `qmk flash` command line if you don't want to edit `rules.mk` (e.g., if you are testing with the `default` keymap before creating your own). There is also the `-bl dfu` option for `qmk flash`, but that option does not recompile the firmware with proper support for that bootloader, so it won't fix any problems with Bootmagic Lite or the `RESET` keycode due to bootloader type mismatch.

Also I did not ever test whether using `nix-shell` would work properly on anything except NixOS x86_64 (but probably the only major difference is the installation of udev rules — the procedure suggested by `qmk doctor` won't work on NixOS, but because you are actually using a more traditional Linux distribution, this won't be a problem for you).

If you still cannot figure out how to fix your problem, please show a screenshot of QMK Toolbox after a successful flash (it would contain information about your controller), or show the `lsusb` output after you reset the controller to the bootloader mode.

kip93, 2022-02-26:
THANK YOU! I am indeed using the Elite-C, and I thought that it was supposed to be a drop in replacement for the pro micro, so I did not think of double checking the bootloader configuration. I just run the commands
```sh
nix-shell --run 'qmk compile -e BOOTLOADER=atmel-dfu -kb crkbd -km default'
nix-shell --run 'qmk flash -e BOOTLOADER=atmel-dfu -kb crkbd -km default'
```
and it worked like a charm.

Closing the issue, and thanks again!

closed 2022-02-26 by kip93