qmk/qmk_firmware #16470: [Feature Request] By holding a key I switch layers and send a special key

ErickTucto, 2022-02-28:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ x] New behavior

## Description

A new function that holding the key down activates a layer and also sends a special key, like KC_LSHFT

sigprof, 2022-02-28:
There are existing keycodes for that purpose: `LM(LOWER, MOD_LSFT)`. One limitation of `LM()` is that only the left side modifiers are supported.

Alternatively, you can easily define a custom keycode for that purpose, and you would be able to use arbitrary keycodes, even ones that are usually not considered to be modifiers. The implementation in `process_record_user()` can be something like this:
```c
case LOWER_RSFT:
    if (record->event.pressed) {
        layer_on(LOWER);
        register_code(KC_RSFT);
    } else {
        unregister_code(KC_RSFT);
        layer_off(LOWER);
    }
    return false;
```

ErickTucto, 2022-02-28:
Thank you very much, I went for the option to use `process_record_user`

closed 2022-02-28 by ErickTucto