qmk/qmk_firmware #16489: [Bug] Drop Alt - QMK configurator generated .bin is bricking the keyboard

skulltraill, 2022-03-01:
## Describe the Bug

After creating a config for drop alt on QMK configurator, compiling, and downloading the `.bin`, and flashing it using QMK toolbox, the keyboard dies, no RGB, no keypresses, it's basically bricked, until I reflash with an older or the default `.bin` which restores it.

This lead me to believe that the `.bin`s from older QMK configurator compilations are working fine, but not the current one on config.qmk.fm

I know the QMK configurator prints out:
`@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
This MCU support package has a lack of support from the upstream provider (Massdrop).
There are currently questions about valid licensing, and at this stage it's likely
their boards and supporting code will be removed from QMK in the near future. Please
contact Massdrop for support, and encourage them to align their future board design
choices to gain proper license compatibility with QMK.
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@`

So this is likely not your fault, nor am I expecting a fix, just wanted to bring this to your attention.

Thank you!

## System Information

 - Keyboard:  Drop Alt
   - Revision (if applicable):  
 - Operating system: Windows 10
 - AVR GCC version: 'avr-gcc' is not recognized as an internal or external command, operable program or batch file.

 - ARM GCC version: 

 - QMK Firmware version: 0.1.1 (I think)

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

zvecr, 2022-03-02:
Can you also provide the following:
* `mdloader` version used
* if eeprom reset was performed?
* `or the default .bin` where was this file acquired from?

zvecr, 2022-03-02:
A quick bisect run from the last develop merge, currently looks like 15953 is the culprit.

PR #16493 (2022-03-03): Fix Drop ALT/CTRL startup issues

## Description

#15953 changed the behaviour of a few core API. With the `arm_atsam` code implementing its own main logic, this has introduced #16489.

This PR fixes the use of ISSI driver config before init,
* `matrix_init` called twice - the first time way too early compared to other platforms
* `issidrv` used before structure configured

Given how the RIOT port has progressed, I didnt see it worth investing a ton of time on the fix.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or upd [... cut]
closed 2022-03-03 by zvecr