qmk/qmk_firmware #16509: sick68 currently uses 0xFEED vid

xia0, 2022-03-03:
cc  @umbynos 

vid is currently set to 0xFEED. Can this please be changed so that a via port can be written?

umbynos, 2022-03-03:
Yeah this can be done, should I generate a random/unique one?

xia0, 2022-03-03:
> Yeah this can be done, should I generate a random/unique one?

I think the recommendation is pick one that might be meaningful to you. From what I have read, the vendor id seems informal and only VIA cares about it not being the default probably because they use the vendor id and product id to identify the device.

umbynos, 2022-03-04:
0x5E68 could be a good fit?

xia0, 2022-03-04:
> 0x5E68 could be a good fit?

Sure

xia0, 2022-03-07:
via keymap PR has been merged (https://github.com/qmk/qmk_firmware/pull/16508)
Just waiting on vendor id change before it can be submitted to the via repository. Would you like for me to submit a PR with the vendor id as 0x5E68?

umbynos, 2022-03-08:
Done!