qmk/qmk_firmware #16521: [Feature Request] Please add support for Unicode input via Compose key on Linux (and BSD) as well

xtaran, 2022-03-04:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Unicode support on Linux currently requires an additional daemon (`ibus-daemon`) with which quite some people are not happy with. The original Unicode input method on Linux (and likely also on BSDs and other Unices) is the Compose key.

QMK already supports Sam Hocevar's WinCompose. But WinCompose is nothing else but a reimplementation of the Unix Compose key feature for Windows. (Sam Hocevar is also a Linux guy. He was Debian Project Leader for at least one term.)

So I wonder if you could add a UC_LNXC (and UC_BSDC as an alias probably) to support Compose key on Linux and BSD (and probably any other X-using Unix platform) as well.

I don't know how close Sam's WinCompose follows the Xorg compose key combinations, but the latter can be found under https://gitlab.freedesktop.org/xorg/lib/libx11/-/blob/master/nls/en_US.UTF-8/Compose.pre — at least those are the ones I expect.

drashna, 2022-03-10:
https://docs.qmk.fm/#/feature_unicode?id=input-modes

Also, qmk allows for configuration of which key(s) to use, as well as to customize the unicode input start and stop. 
https://docs.qmk.fm/#/feature_unicode?id=additional-customization

xtaran, 2023-03-29:
> This issue has been automatically closed because it has not had activity in the last 30 days. If this issue is still valid, re-open the issue and let us know.

It's a feature request. Of course it is still valid. \*sigh\*/ But I don't seem to be able to reopen it…

tzarc, 2023-03-29:
If we left all feature requests perpetually open, we wouldn't be able to make sense of things.
If they're active they can stay open, in the hope that someone produces a PR.
If there isn't enough interest or motivation to trigger someone to take this through to completion, then that's indication enough that it isn't a high enough priority.

We have limited resources -- if you wish to use yours to get this over the line, feel free.

Reopening -- if stalebot kicks in without progress again I'll just short-circuit the process and close this issue.

closed 2023-06-28 by tzarc