qmk/qmk_firmware #16553: redox_w firmware broke between 0.15.25 and 0.16

achatham, 2022-03-07:
I could no longer produce a working firmware via configurator, while it worked before. I've built maually and verified that it broke between 0.15.25 (works) and 0.16 (does not work), and head also doesn't work. I've tried git bisect but must not be doing it right yet. I'll keep searching to try to figure out the cause.

The bad firmware flashes successfully but no keypresses register.

achatham, 2022-03-07:
Confirmed it's from c12268807d8622a05dc445e6101a575eae16860d, which changed UART. I'm on windows 11 and the receiver being flashed has a USB-C connection.

achatham, 2022-03-07:
Based on the variables, I'm guessing this is about USB-C and not redox_w. But I'm at my limit of being able to debug the next step, as I'm not too familiar with serial comms.

drashna, 2022-03-10:
@fauxpark any insight?

tomaszj, 2022-05-12:
I can confirm this — compiling firmware on `0.15.25` works as expected. 

`0.16.9` tag and `master` branch locally are broken, whether compiled on a MacBook Air or via Docker — I have tried both routes.

Using firmware from QMK Configurator suffers from the exact same issue.

Note that I can get keys fine **after** erasing EEPROM and then flashing. Keys stop being sent after a single reset.

cc @mattdibi

danielmitchell, 2022-05-25:
I'm also experiencing this issue and have been unable to update my firmware using QMK Configurator since the issue was introduced. Thanks for addressing the issue since my layout is causing RSI and needs to be updated. Hoping this can be fixed soon.

achatham, 2022-05-25:
I'm curious what the common thread is. Do others experiencing this have a
usb-c connection? A redox-w?

danielmitchell, 2022-05-25:
Yes, I'm using a usb-c connection with a redox-w

mffseal, 2022-06-04:
I have the same problem.

closed 2022-06-07 by drashna

tomaszj, 2022-06-07:
The fix is now in `master` @achatham @danielmitchell @mffseal

PR #17203 (2022-06-07): Handle timeout on UART for Redox Wireless receiver-to-keyboard communication.
- This fixes the issue of keyboard deadlocking on the first matrix
  scan with Redox Wireless keyboards

## Description

Resolves #16553. Closes #17071 

This brings back handling time-out in matrix scanning code. Versions `0.15.x` and before handled timeout implicitly, by using a non-blocking `UDR` register read.

The timeout variable is shared across each matrix scan — happy to change it to be individual to each UART byte read instead on a clear recommendation from someone more knowledgeable in UART comms.

The cur [... cut]