qmk/qmk_firmware #16555: [Bug] Various errors trying to compile dztech/dz60rgb_ansi/v1 in freebsd

toniz4, 2022-03-07:
When trying to compile the dztech/dz60rgb_ansi/v1:default keyboard layout, it fails to build with the following errors:

```
Compiling: lib/chibios/os/various/syscalls.c
   lib/chibios/os/various/syscalls.c:71:20: warning: 'struct _reent' declared inside parameter list will not be visible outside of this definition or declaration
 int _read_r(struct _reent *r, int file, char * ptr, int len) {
                    ^~~~~~
lib/chibios/os/various/syscalls.c: In function '_read_r':
lib/chibios/os/various/syscalls.c:84:3: warning: implicit declaration of function '__errno_r'; did you mean '__error'? [-Wimplicit-function-declaration]
   __errno_r(r)  = EINVAL;
   ^~~~~~~~~
   __error
lib/chibios/os/various/syscalls.c:84:17: error: lvalue required as left operand of assignment
   __errno_r(r)  = EINVAL;
                 ^
lib/chibios/os/various/syscalls.c: At top level:
lib/chibios/os/various/syscalls.c:92:21: warning: 'struct _reent' declared inside parameter list will not be visible outside of this definition or declaration
 int _lseek_r(struct _reent *r, int file, int ptr, int dir) {
                     ^~~~~~
lib/chibios/os/various/syscalls.c:104:21: warning: 'struct _reent' declared inside parameter list will not be visible outside of this definition or declaration
 int _write_r(struct _reent *r, int file, char * ptr, int len) {
                     ^~~~~~
lib/chibios/os/various/syscalls.c:121:21: warning: 'struct _reent' declared inside parameter list will not be visible outside of this definition or declaration
 int _close_r(struct _reent *r, int file) {
                     ^~~~~~
lib/chibios/os/various/syscalls.c:131:24: warning: 'struct _reent' declared inside parameter list will not be visible outside of this definition or declaration
 caddr_t _sbrk_r(struct _reent *r, int incr) {
                        ^~~~~~
lib/chibios/os/various/syscalls.c: In function '_sbrk_r':
lib/chibios/os/various/syscalls.c:139:19: error: lvalue required as left operand of assignment
     __errno_r(r)  = ENOMEM;
                   ^
lib/chibios/os/various/syscalls.c: At top level:
lib/chibios/os/various/syscalls.c:153:21: warning: 'struct _reent' declared inside parameter list will not be visible outside of this definition or declaration
 int _fstat_r(struct _reent *r, int file, struct stat * st) {
                     ^~~~~~
lib/chibios/os/various/syscalls.c:165:22: warning: 'struct _reent' declared inside parameter list will not be visible outside of this definition or declaration
 int _isatty_r(struct _reent *r, int fd) {
                      ^~~~~~
 [ERRORS]
 |
 |
 |
gmake[1]: *** [builddefs/common_rules.mk:456: .build/obj_dztech_dz60rgb_ansi_v1/lib/chibios/os/various/syscalls.o] Error 1
```

qmk doctor:
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /usr/home/cassio/src/qmk_firmware-valilla
⚠ Unsupported OS detected: freebsd-13.0-stable-amd64-64bit-elf
Ψ Git branch: master
Ψ Repo version: 0.16.3
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.4.0
Ψ Found avr-gcc version 10.3.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

rafiks, 2022-06-28:
any resolution to this? I am seeing the same error.

tzarc, 2022-06-28:
Unfortunately not -- you can see from the `qmk doctor` output that FreeBSD is a largely unsupported build platform for QMK.
We've always been reliant on the community to provide fixes -- at this stage someone with a bit more familiarity will need to step up and provide a fix.

rafiks, 2022-06-28:
> 
i  see that on the QMK doctor.. I think you should remove Freebsd from the setup page or at the very least put a warning not to rely on FreeBSD. https://docs.qmk.fm/#/newbs_getting_started

closed 2022-08-29 by tzarc