qmk/qmk_firmware #16565: kbdfans/kbd67/mkiirgb_iso does not compile - check-size fails

emilerl, 2022-03-07:
## Describe the Bug

Compiling mkiirgb_iso with the default keymap yelds an error on check-size (see output below). The example provided below is from the QMK configurator when just selecting `kbdfans/kbd67/mkiirgb_iso` and `LAYOUT_65_iso_blocker` as layout, with no further modifications.

```
Copying kbdfans_kbd67_mkiirgb_iso_layout_65_iso_blocker_mine.bin to qmk_firmware folder             
Checking file size of kbdfans_kbd67_mkiirgb_iso_layout_65_iso_blocker_mine.bin                     
 *  24800/24576 (224 bytes over)
 ```

I saw that v3 of mkiirgb defines `BOOTLOADER_SIZE = 6144`, but I'm not entirely sure if that setting is applicable for the iso version?

### qmk doctor

Although the example above is from the QMK configurator, I've reproduced the same issue locally on my MacBook with the following output:

```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /Users/emil/qmk_firmware
Ψ Detected macOS 12.0.1.
Ψ Git branch: master
Ψ Repo version: 0.16.0
⚠ Git has unstashed/uncommitted changes.
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

waffle87, 2022-03-09:
Something from here should definitely put you under the size limit 
[https://docs.qmk.fm/#/squeezing_avr](https://docs.qmk.fm/#/squeezing_avr)

> I saw that v3 of mkiirgb defines `BOOTLOADER_SIZE = 6144`, but I'm not entirely sure if that setting is applicable for the iso version?
It is, as the lufa-ms bootloader is still used 
[https://github.com/qmk/qmk_firmware/blob/master/keyboards/kbdfans/kbd67/mkiirgb_iso/rules.mk#L5](https://github.com/qmk/qmk_firmware/blob/master/keyboards/kbdfans/kbd67/mkiirgb_iso/rules.mk#L5)

emilerl, 2022-03-09:
Thank you @waffle87, `LTO_ENABLE = yes` did the trick for now.

emilerl, 2022-03-10:
@waffle87, I tried recompiling with the changes introduced and can confirm that `BOOTLOADER_SIZE = 6144` gives a problem. With that set, I'm not able to soft reset the PCB to flash, I have to use the hardware reset button on the back of the PCB. I tried just removing this parameter again, and then it works as expected.

drashna, 2022-03-10:
Closing, as this is also merged into master.

closed 2022-03-10 by drashna