qmk/qmk_firmware #16581: [Bug] Shifted keys not working with modifier keys in german layout

Thieso, 2022-03-08:
## Describe the Bug

I would like to use home row mods with the german keyboard layout. For this I have the following mapping which should give a semicolon when pressed and released and the left gui key on hold. 

```
LGUI_T(DE_SCLN)
```
Sadly this does not produce a semicolon but only a comma when pressed and released. Am I doing something wrong or is this a bug?
\DE_SCLN\ is defined as `S(KC_COMM)` so it seems like the shift is getting ignored but why?

## System Information

**Keyboard:** Ferris sweep
**Operating system:** Ubuntu 20.04
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/thies/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/thies/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/thies/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32' boards. Run 'sudo cp /home/thies/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/thies/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/thies/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'massdrop' boards. Run 'sudo cp /home/thies/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/thies/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/thies/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.16.4
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ no] AutoHotKey (Windows)
 - [ no] Karabiner (macOS)
 - [ no] Other:

## Additional Context

fauxpark, 2022-03-08:
https://docs.qmk.fm/#/mod_tap?id=caveats

Thieso, 2022-03-08:
Ah thanks for the quick reply. That solves it

closed 2022-03-08 by Thieso