qmk/qmk_firmware #16582: TM680 Single Mode Lock issue

Frosty680, 2022-03-08:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Hi,
Idk if this is the right website to submit, but can I request a new firmware for TM680 Single Mode, it has a bug or issue that it will lock your keyboard, you cannot use win and fn key and the knob and side LED is stuck to color purple/violet.
I hope you can help me if this is the right page.

Thank you :)

zvecr, 2022-03-08:
Sorry but this is not a QMK supported keyboard.

closed 2022-03-08 by zvecr

Frosty680, 2022-03-09:
Alright, thank you.

bintuananh1107, 2022-03-25:
> Alright, thank you.

can you help me, my keyboard is the same as yours

bintuananh1107, 2022-03-25:
> Alright, thank you.

email me , bintuananh@gmail.com