qmk/qmk_firmware #16622: qmk支持模拟surface dial吗Qmk zhīchí mónǐ surface dial ma, 21 / 5,000 翻译结果 Does qmk support simulating surface dial?

jeromeli002, 2022-03-12:
qmk支持模拟surface dial吗，看见有人实现了不知道qmk有没有原生支持？有键盘支持的开源参考吗

Does qmk support simulating surface dial? I saw that someone has implemented it. I wonder if qmk has native support? Is there an open source reference for keyboard support?

7-liang, 2022-03-16:
他那个放到屏幕上即可在屏幕上弹出菜单，这个应该和屏幕之间有某种近场感应部件的存在，这个就和qmk无关了，需要你自行研究他的触发方式，并且他弹出的菜单，你选择某一项后，他的向左转，向右转的功能都不一样

用qmk实现个类似的倒是没问题的，但完全替代，不太可能吧

jeromeli002, 2022-03-16:
> 他那个放到屏幕上即可在屏幕上弹出菜单，这个应该和屏幕之间有某种近场感应部件的存在，这个就和qmk无关了，需要你自行研究他的触发方式，并且他弹出的菜单，你选择某一项后，他的向左转，向右转的功能都不一样
> 
> 用qmk实现个类似的倒是没问题的，但完全替代，不太可能吧
要能按下弹出菜单就行，设备要多个轮盘出来现在识别只有键盘

7-liang, 2022-03-16:
首先，咱也没有苏菲，其次，弹出菜单这个功能是人家的商业机密，需要你自行去研究怎么把它实现出来了，估计你在这是问不出所以然的，除非有人已经实现了，你可以去打听对方怎么实现的
那种在旋钮周围弹出的菜单，应该是内部的一个机制，近场感应到那玩意的存在，自动弹出菜单，如果这个功能有快捷键可以调出，那么，qmk即可实现
首先，你需要买一个这货回来，然后把他大卸八块，然后再逐个研究

jeromeli002, 2022-03-16:
> 首先，咱也没有苏菲，其次，弹出菜单这个功能是人家的商业机密，需要你自行去研究怎么把它实现出来了，估计你在这是问不出所以然的，除非有人已经实现了，你可以去打听对方怎么实现的 那种在旋钮周围弹出的菜单，应该是内部的一个机制，近场感应到那玩意的存在，自动弹出菜单，如果这个功能有快捷键可以调出，那么，qmk即可实现 首先，你需要买一个这货回来，然后把他大卸八块，然后再逐个研究

有人实现了的 所以qmk是可以改改支持的，有个app也能让手机蓝牙也能识别出轮盘出来

7-liang, 2022-03-16:
> 抄　送：李钧 ***@***.***>; Comment ***@***.***>
> 主　题：Re: [qmk/qmk_firmware] qmk支持模拟surface dial吗Qmk zhīchí mónǐ surface dial ma, 21 / 5,000 翻译结果 Does qmk support simulating surface dial? (Issue #16622)

按你这么说的话，应该是有键码可以调出那个界面，你得先去打听到
有了键码就好办了

--------------------------------------------------------------------------
 顺 祝 商 祺
--------------------------------------------------------------------------
李  钧

------------------------------------------------------------------

closed 2022-08-27 by github-actions[bot]