qmk/qmk_firmware #16631: [Bug] Making crkbd/rev1 with keymap default iv

geedmo, 2022-03-13:
## Describe the Bug

## System Information

**Keyboard:** 
**Revision (if applicable):** 
**Operating system:** 
**`qmk doctor` output:** 
```
(Paste output here)
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

zvecr, 2022-03-14:
`qmk compile -kb crkbd/rev1 -km default -e RGB_MATRIX_ENABLE=yes -e RGBLIGHT_ENABLE=no`

Outputs
` * The firmware size is fine - 24626/28672 (85%, 4046 bytes free)`

geedmo, 2022-03-15:
Thanks. Did not notice and I thought it would not be necessary since `RGBLIGHT_ENABLE=no` is  already declared in rules.mk

zvecr, 2022-03-15:
https://github.com/qmk/qmk_firmware/blob/7fde30979986732433126d50911efdef928c2b64/keyboards/crkbd/keymaps/default/rules.mk#L2

closed 2022-03-15 by geedmo