qmk/qmk_firmware #16648: RESET vs QK_BOOT?

mcp292, 2022-03-14:
There seems to be two documented ways to put the keyboard in bootloader
mode:
1.
[`RESET`](https://docs.qmk.fm/#/ref_functions?id=resetting-the-keyboard).
2. [`QK_BOOT`](https://docs.qmk.fm/#/quantum_keycodes?id=qmk-keycodes).

Which is the better one to use, what are the differences?

Similarly, `EEP_RST` vs `EE_CLR`?

KarlK90, 2022-03-14:
`QK_BOOT` or `QK_BOOTLOADER` is the recommended way. Same is true for `EE_CLR`.

mcp292, 2022-03-14:
Thank you. Is this documented somewhere that I missed?

mcp292, 2022-03-17:
Ah, okay thanks!

@fauxpark based on my experience with these changes, do you think it worthwhile to document these as the new standard and remove [the keycodes they replace in the documentation](https://docs.qmk.fm/#/ref_functions?id=resetting-the-keyboard) with a note.

fauxpark, 2022-03-18:
Not sure how I missed those in the above PR... I know Configurator also needs to be updated to recognise these new names.

mcp292, 2022-03-18:
Okay great!

closed 2022-07-04 by drashna