qmk/qmk_firmware #16651: Add support for globe key (🌐︎) for macOS/iPadOS/iOS

adiabatic, 2022-03-14:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Now that [Universal Control](https://www.macrumors.com/2022/03/14/apple-releases-macos-monterey-12-3/) has been released (in beta) for macOS and iPadOS, I'm getting a chance to use my Ergodox EZ on more than just macOS. However, recent versions of iPadOS (and macOS) use a new modifier button — the globe key (🌐︎). On older Mac keyboards, this functionality is shoved onto the fn key, down in the lower left corner of the keyboard.

At any rate, having a binding for the globe key on macOS is somewhat useful. If your Mac has one, then the keyboard shortcut for `Full Screen` will be the simple 🌐︎F instead of ⇧⌘F or ⌥⌘F or whatever it is depending on the application. On iPadOS, having a globe key is even more useful — 🌐︎S will open Siri, 🌐︎← will show the previous application, etc.

However, when browsing https://docs.qmk.fm/#/keycodes, I don't see any "globe" or "fn" key listed. Any idea if it'd be possible to add this sort of key to `qmk_firmware`? I ask, in part, because I've never seen support for Apple's fn keys — which might mean it's some kind of never-before-been-emulated keycode that nobody's been able to tap into, see, and emulate.

pyrho, 2022-03-21:
See https://github.com/qmk/qmk_firmware/issues/2179 for the discussion and [AppleFn.patch](https://gist.github.com/fauxpark/010dcf5d6377c3a71ac98ce37414c6c4) for the actual patch. (I guess you can close this)

adiabatic, 2022-03-21:
Hm, pity this issue isn't getting linked from #2179. I'll keep it open just in case it's implemented differently, but I doubt it.

closed 2022-09-04 by github-actions[bot]

stanly0726, 2023-02-19:
any solution?

adiabatic, 2023-02-19:
In Ventura, you can turn the otherwise-useless caps lock key into an fn/globe key:
[image]

ultrasiq, 2023-02-28:
> In Ventura, you can turn the otherwise-useless caps lock key into an fn/globe key:
> [image]

THANK YOU

HVR88, 2023-10-11:
Check page 76 ->  Table 13-3 HID Consumer Page (0x0C) controls for use by keyboards

https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

0x029D. AC Keyboard Layout Select. ---> Globe Key

luohda, 2024-01-24:
> Check page 76 -> Table 13-3 HID Consumer Page (0x0C) controls for use by keyboards
> 
> https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf
> 
> 0x029D. AC Keyboard Layout Select. ---> Globe Key

just used this method to flash to my nuphy keyboard, works wonderfully! thanks!

adiabatic, 2024-01-25:
<https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf#page=86> if you don't want to press ⌘F to search for `029D`.

adiabatic, 2024-08-25:
Things start to get interesting for us at <https://gist.github.com/fauxpark/010dcf5d6377c3a71ac98ce37414c6c4?permalink_comment_id=4753144#gistcomment-4753144>. Apparently, the behavior is rather buggy.

Personally, I think I'm going to stick with caplock-as-globe — I use my usual layout in Linux sometimes, and capslock there works great as a _compose_ key, whereas a globe key would be totally useless.