qmk/qmk_firmware #16675: "TIM2 is not a 32bits timer when compile

thzx805, 2022-03-17:
when compile met this error as below

 lib/chibios/os/hal/ports/STM32/LLD/TIMv1/hal_st_lld.c:48:2: error: #error "TIM2 is not a 32bits timer"
   48 | #error "TIM2 is not a 32bits timer"

make[1]: *** [builddefs/common_rules.mk:456：.build/obj_Henry/lib/chibios/os/hal/ports/STM32/LLD/TIMv1/hal_st_lld.o] 错误 1
Make finished with errors

**Ruls.mk**

/  # MCU name
MCU = STM32F103
 / #MCU_LDSCRIPT = stm32f103_bootloader
 / #DFU_ARGS = -d 0483:df11 -a 0 -s 0x08000000:leave
/  # Bootloader selection
  BOOTLOADER = stm32duino 
   BOARD = STM32_F103_STM32DUINO

/  # Build Options
 / #   change yes to no to disable
/  #
BOOTMAGIC_ENABLE = no       # Enable Bootmagic Lite
MOUSEKEY_ENABLE = yes       # Mouse keys
EXTRAKEY_ENABLE = yes       # Audio control and System control
CONSOLE_ENABLE = yes        # Console for debug
COMMAND_ENABLE = yes        # Commands for debug and configuration
NKRO_ENABLE = no            # Enable N-Key Rollover
BACKLIGHT_ENABLE = no       # Enable keyboard RGB underglow
AUDIO_ENABLE = no           # Audio output

LAYOUTS = ortho_5x5

 |

ritaru, 2022-04-06:
Check the below issue.

https://github.com/qmk/qmk_firmware/issues/13816

WillD7545, 2022-04-17:
> Check the issue below.
> 
> #13816

This is kind of annoying as MSYS doesn't build a chconfig.h when going through the new keyboard setup.

zvecr, 2022-04-21:
Defaults updated by #https://github.com/qmk/qmk_firmware/pull/16909.

Otherwise refer to the the suggestion within #13816.

PR #16909 (2022-04-21): Provide better config defaults for bluepill boards

## Description

Also produces a better out of box experience for `qmk new-keyboard`.

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #16675

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [ [... cut]
closed 2022-04-21 by zvecr