qmk/qmk_firmware #16683: [Bug] Mouse movement freaks out on resume from suspend - on chibiOS

drashna, 2022-03-18:
## Describe the Bug

when resuming from sleep/suspend, the mouse movement (x and y reports) don't work.  with my large mouse reports, it does... kind of, but much slowed down. To the point of unusable. 

Similarly odd behavior happens with mousekeys. 

## System Information

**Keyboard: Tractyl Manuform 5x6 blackpill (f411)
**Operating system:** macOS
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /Users/drashna/repos/qmk_firmware
Ψ Detected macOS 11.6.4.
Ψ Git branch: develop
Ψ Repo version: 0.16.5
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.2.1
Ψ Found avr-gcc version 11.1.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

manna-harbour, 2022-09-25:
Might be related to https://github.com/qmk/qmk_firmware/issues/16406.

drashna, 2022-10-07:
Likely.  It's definitely an issue with the usb endpoint, and the report sending.  extended mouse reports seem to work better, but still have issues, for instance.

drashna, 2023-05-11:
`USB_SUSPEND_WAKEUP_DELAY` seems to help.