qmk/qmk_firmware #16698: [Bug] Mouse speed sped up after git pull and flash

mcp292, 2022-03-20:
Somewhere in-between 669ad3c805 and d175ee1d9b something changed the
mouse speed settings, making what were slow parameters, result in fast
movement. This bug could be restricted to [constant
mode](https://docs.qmk.fm/#/feature_mouse_keys?id=constant-mode); I
haven't had time to test further.

Here are my settings:
```txt
config.h
----
#define MK_3_SPEED
#define MK_MOMENTARY_ACCEL

#define MK_C_OFFSET_1   1       /* Cursor offset per movement (KC_ACL1) */
#define MK_C_INTERVAL_1 10      /* Time between cursor movements (KC_ACL1) */
#define MK_W_OFFSET_1   1       /* Scroll steps per scroll action (KC_ACL1) */
#define MK_W_INTERVAL_1 101	/* Time between scroll steps (KC_ACL1) */

#define MK_C_OFFSET_UNMOD   1   /* Cursor offset per movement (UNMOD) */
#define MK_C_INTERVAL_UNMOD 0   /* Time between cursor movements (UNMOD) */
#define MK_W_OFFSET_UNMOD   1   /* Scroll steps per scroll action (UNMOD) */
#define MK_W_INTERVAL_UNMOD 101	/* Time between scroll steps (UNMOD) */

#define MK_C_OFFSET_2   5       /* Cursor offset per movement (KC_ACL2) */
#define MK_C_INTERVAL_2 0       /* Time between cursor movements (KC_ACL2) */
#define MK_W_OFFSET_2   1       /* Scroll steps per scroll action (KC_ACL2) */
#define MK_W_INTERVAL_2 50      /* Time between scroll steps (KC_ACL2) */
```

```txt
rules.mk
----
MOUSEKEY_ENABLE = yes
```

Just from searching through commit messages I see 0880850b15 touches
some mouse stuff.

On Arch Linux using DWM as the window manager. Keyboard: Fortritude60.

mcp292, 2022-05-26:
Still persists on `f5d091a9d5 * origin/master Small update to Symmetric70 keyboard readme (#17188)`.

mcp292, 2022-05-26:
```c
#define MK_C_OFFSET_1   1       /* Cursor offset per movement (KC_ACL1) */
#define MK_C_INTERVAL_1 20      /* Time between cursor movements (KC_ACL1) */
#define MK_W_OFFSET_1   1       /* Scroll steps per scroll action (KC_ACL1) */
#define MK_W_INTERVAL_1 200   	/* Time between scroll steps (KC_ACL1) */

#define MK_C_OFFSET_UNMOD   1   /* Cursor offset per movement (UNMOD) */
#define MK_C_INTERVAL_UNMOD 10  /* Time between cursor movements (UNMOD) */
#define MK_W_OFFSET_UNMOD   1   /* Scroll steps per scroll action (UNMOD) */
#define MK_W_INTERVAL_UNMOD 101	/* Time between scroll steps (UNMOD) */

#define MK_C_OFFSET_2   1       /* Cursor offset per movement (KC_ACL2) */
#define MK_C_INTERVAL_2 1       /* Time between cursor movements (KC_ACL2) */
#define MK_W_OFFSET_2   1       /* Scroll steps per scroll action (KC_ACL2) */
#define MK_W_INTERVAL_2 50      /* Time between scroll steps (KC_ACL2) */
```

This is what I'm set at now. It's as if the interval unit has changed.

mcp292, 2022-06-04:
Pulling master at 787165718d now results in the mouse moving one step per key press, that's it. This is the case with both my original settings and the newly posted.

drashna, 2022-06-21:
do you know which commit specifically is causing the issue?

filterpaper, 2022-06-21:
The last working commit for mousekey.c is 63646e8906e062d1c1de3925cba70c4e3426a855 and 921b9dad6c37575215231b34a3492ffb38eaeec2 broke the constant mode.

drashna, 2022-06-21:
That's ... very odd, and shouldn't have impacted anything.

Could you try adding this to your code, and see if that fixes your issue? 
```c
bool has_mouse_report_changed(report_mouse_t* new_report, report_mouse_t* old_report) {
    return (mouse_report.x || mouse_report.y || mouse_report.v || mouse_report.h);
}
```

filterpaper, 2022-06-21:
> That's ... very odd, and shouldn't have impacted anything.
>
> Could you try adding this to your code, and see if that fixes your issue?
>
> ```c
> bool has_mouse_report_changed(report_mouse_t* new_report, [...]

This fixes the issue and also explained the oddity with kinetic mouse wheel.

filterpaper, 2022-06-29:
#17493 should have fixed this

Markus00000, 2022-06-29:
I can confirm that the “one step per key press” issue is fixed.

mcp292, 2022-06-29:
> I can confirm that the “one step per key press” issue is fixed.

Same here. Thanks for the hard work boys! 

There definitely is still a speedup as discussed, but mouse functionality is effectively restored. I defer to your judgement as to whether or not to close this issue.

closed 2022-07-04 by drashna