qmk/qmk_firmware #16700: doc mistake

tara81, 2022-03-20:
https://docs.qmk.fm/#/feature_stenography?id=plover-with-steno-protocol

void eeconfig_init_user() {
    steno_set_mode(STENO_MODE_GEMINI); // or STENO_MODE_BOLT
}

should be 

void matrix_init_user() {
steno_set_mode(STENO_MODE_GEMINI);
}

MacroLens, 2022-04-05:
This issue may be incorrect, as I don't think the comment is incorrect. It's well documented that you can set the stenography mode to either Gemini or Bolt. What exactly is the issue? If there is an actual issue with STENO_MODE_BOLT can you detail it?

tara81, 2022-04-05:
When I set steno to Gemini , it defaults to bolt. Like the documentation is wrong?

tzarc, 2022-04-06:
It's not wrong, it sets it once and only once, when the EEPROM is reset. That is the purpose of the `eeconfig_xxxxxxx` class of function. After that, whatever is selected by the user is retained as the default.

At some point, `STENO_MODE_BOLT` must have been stored in EEPROM. The next powerup, it reloads `STENO_MODE_BOLT`.

precondition, 2022-05-15:
> When I set steno to Gemini , it defaults to bolt. Like the documentation is wrong?

@tara81 Some keyboards don't support EEPROM, in which case your choice isn't memorised and you'll need to press `QK_STENO_GEMINI` every time you reboot your keyboard in order to select GeminiPR. That limitation only affects a few QMK keyboards (not sure of the exact proportion) but if you're in this situation, you may be interested in [my steno PR](https://github.com/qmk/qmk_firmware/pull/17065) which doesn't rely on EEPROM and lets you compile only your desired steno protocol.

closed 2022-09-13 by github-actions[bot]