qmk/qmk_firmware #16705: [Bug] rgb matrix Digital rain ignores RGB_MATRIX_MAXIMUM_BRIGHTNESS

KoenGoe, 2022-03-21:
The digital rain rgb matrix animation has its own hardcoded constants for maximum brightness, instead of using `RGB_MATRIX_MAXIMUM_BRIGHTNESS`, nor is checked it in `rgb_matrix_set_color` that it calls.
So it is checked nowhere, and leds are just set at brightnesses over the maximum set by `RGB_MATRIX_MAXIMUM_BRIGHTNESS`, which is potentially dangerous as it can lead to too much current drawn from the USB port.

**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: D:/koeng/Documents/keyboard/qmk
Ψ Detected Windows 10 (10.0.19044).
Ψ Git branch: master
Ψ Repo version: 0.15.23
⚠ Git has unstashed/uncommitted changes.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

drashna, 2022-03-21:
Can confirm.   Not sure there is an easy way to fix this, but this line may be what needs to be modified: 
https://github.com/qmk/qmk_firmware/blob/7eb6f86bc0aac3ff83abe4365cd11c5c195dc403/quantum/rgb_matrix/animations/digital_rain_anim.h#L15

setting that to the current val instead of 0xFF may work, but the other colors may not be correct. In fact, they may need to be calculated at run time.

KoenGoe, 2022-03-22:
Yeah I tried these settings:
```
    const uint8_t pure_green_intensity = RGB_MATRIX_MAXIMUM_BRIGHTNESS*3/4;
    const uint8_t max_brightness_boost = RGB_MATRIX_MAXIMUM_BRIGHTNESS*3/4;
    const uint8_t max_intensity        = RGB_MATRIX_MAXIMUM_BRIGHTNESS;
```
So that the constants have about the same ratio as originally.
While this kept the brightness under `RGB_MATRIX_MAXIMUM_BRIGHTNESS` is still has two issues:
 * it's ignoring the current `val`. At least it's an electrically safe setting now, but of course it would be better if the effect brightness followed the `val` as well
 * The decay rate is 1 (`--`) per update and should be slowed down inversely proportional to `max_intensity`.

commit 671b62ef (2022-04-19): Fixes #16705 : digital rain follows val (#16716)
closed 2022-03-24 by KoenGoe