qmk/qmk_firmware #16772: Support for Ducky One 2 Mini 2061ST

deadbeef2000, 2022-04-02:
Hello,

I haven't tried installing the Ducky One 2 Mini firmware for the 1861ST yet, but I wanted to make sure that I won't break my keyboard either. Does anyone have any advice on this or tried this already?

dscain, 2022-04-03:
I have tried 1861ST and it doesn't work. It will not brick, but you will need to re-flash with the official Ducky flash tool to bring the keyboard back into a usable state. https://www.duckychannel.com.tw/en/Support/Detail/19

firatakandere, 2022-06-14:
Unfortunately 2061ST has M032SE3AE chipset, which is different than the other Ducky One 2 Mini.

tzarc, 2022-08-29:
This would need to start with adding support to ChibiOS-Contrib for that MCU.
QMK-side can happen afterwards -- closing until that occurs.

closed 2022-08-29 by tzarc

timfenney, 2023-04-07:
I might be missing something here, but according to this site, all mcus with a cortex M0 are supported:

http://www.chibios.org/dokuwiki/doku.php?id=chibios:documentation:books:rt:port_armv6m

tzarc, 2023-04-07:
It'll run on an M0/M0+. But there's no peripheral definitions, so whilst you can execute code, it can't talk over USB, can't manipulate GPIO in order to scan a matrix, and can't access any other peripherals such as I2C- or SPI-based peripherals like OLEDs and the like.

It's like having a functional brain but no body to work with.

rRobis, 2023-09-29:
Is there any real possibility, that this would become a reality? I really don't want to buy an identical keychronic, just to get QMK, but i really would love to have it.

tzarc, 2023-09-29:
Real possibility? Not really.
For even one chip to work you'd need to implement a whole family's worth of code, and only supply parameters for a single chip.

Refer to these existing locations for an example -- **everything** WB32 would need an equivalent to be written for the M032 family:
https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-21.11.x/os/common/ext/CMSIS/WB32
https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-21.11.x/os/common/startup/ARMCMx/devices
https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-21.11.x/os/hal/ports/WB32

I'm sure there are other locations, but you're looking at months of work.

rRobis, 2023-09-30:
Thanks for the answer.

I guess I have to figure out how to buy an ergo split keyboard in Europe then, for me to make it worth the upgrade.