qmk/qmk_firmware #16795: [Bug report] Keys override: When supressing modifiers like alt or WIN, it causes those modifiers to immediately fire and break focus

ujl123, 2022-04-04:
## Describe the Bug
For an example I've set LGUI+ 1 = x. I've also set it to suppress the mod LGUI as I don't want to send through LGUI+x because that is the short cut for quick link menu in windows.

mod = LGUI
Trigger = 1
Replacement: x
suppress mod = LGUI
Expect behavior: the letter x being typed
Actual behavior: Types the letter x in vs code, but immediately opens the start menu 

Video:

https://user-images.githubusercontent.com/13132548/161632719-27441447-7a0d-4dd8-b7d0-ab8a6a2eb202.mp4

I used LGUI in the above example because it's the most noticeable but I believe this happens with all modifiers.

Another example below to further illustrate the issue

mod = LAlt
Trigger = 1
replacement = f
suppress mod = LAlt
expected behavior = the letter f being typed in VS code (windows)
Actual behaviour = it opening the menu bar, choosing **F**ile then immediately closing the menu.

Video:

https://user-images.githubusercontent.com/13132548/161632738-d80ba9e6-ded9-45ba-b7d0-f600a78ecbb2.mp4

From observing the effects and talking to others, I suspect that when you use suppress mod it causes the mod key to be release before firing the replacement key? If that's the case using alt in any sort of application that has a menu bar will cause it to either expand or even access one of the menu items if the replacement key is a hotkey one of the menu items.  And when you supress the LGUI key it will always try and activate the start bar.

Is the above fixable or is the solution to caution the users in the documentation on this?

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

ujl123, 2022-04-06:
Some thoughts will thinking about a solution to this issue. The issue stems from the fact that when you choose to suppress the mods, it will release the modifier and then fire off the replacement. The problem is that some modifiers cause unintended effects. For the window key it will open the start bar, and for alt it will trigger the menu in a browser or text editor. 

It seems that if you hit WIN + another modifier, it seems to prevent the default behavior of WIN or ALT. perhaps a solution is when ALT or WIN is being suppressed by itself to introduce another unused modifier to prevent it's default behavior?

For example

mod = LGUI
Trigger = 1
Replacement: x
suppress mod = LGUI
Proposed fix: Release the LGUI key but also add another modifier like CTRL
outcome: Types X, no start bar gets opened.

precondition, 2022-05-03:
It would have to be a user-configurable dummy keycode to insert because there isn't a keycode that's guaranteed to no-op when combined with a modifier, universally on all OS's

ujl123, 2022-05-03:
@precondition  funny you bring that up. I was reading your (i think) blog a few weeks ago mulling over this issue and came across this :

> KC_LANG6 or KC_CRSEL, or a dummy custom keycode created solely for this purpose. Regardless of what you choose, the OS will never receive the dummy keycode from your keyboard.

I haven't had time to look into this yet, but when I have time I might try this out to see if using a dummy keycode will "fix" the GUI or ALT modifier supression

precondition, 2022-05-03:
The idea is good in that it will effectively prevent flashing modifiers and thus avoid opening the start menu or selecting the menus of a graphical app. I was just saying that we can't just hardcode a certain dummy keycode in the core of QMK, it'd have to be something that users can change in `config.h`. We can imagine an option like `KEY_OVERRIDE_DUMMY` that's set to `KC_NO` by default and an user can configure with `#define KEY_OVERRIDE_DUMMY KC_RALT` or `#define KEY_OVERRIDE_DUMMY KC_INT3`.  It can't be `KC_CRSEL` or `KC_LNG6` though because those keycodes aren't even supported by any modern OS. We still need to use something that OS can at least interpret.

ujl123, 2022-05-03:
Thank you for clarifying @precondition ! I might give this a try over the weekend, however I'm not really familiar with C and hopefully someone can actually be assigned to implement this.

precondition, 2022-05-15:
> Edit: Dumb question, but could I just send KC_NO, rather than a custom keycode that is defined as KC_NO?

As far as I know, KC_NO is a no-op so nothing gets sent down the wire which means it would be no different from the actual implementation.

ujl123, 2022-05-25:
Looking at how other programs implement this, it looks like microsoft powertoys uses a left mouse click + something called OemClear to get around the modifier being activated when suppressed. Here's a video on an unrelated ticket, but you can see on the bottom left overlay that when I use a disabled hotkey (lgui + e) it instead does lbutton + OemClear to get ride of the keypress perhaps something similar could be used?

https://user-images.githubusercontent.com/13132548/170129123-7314910c-3b1a-4740-bbd1-f93655376925.mp4

wrinkledeth, 2022-10-24:
@ujl123 did you ever find a solution to this? I came across this same bug when trying to create a combo for WIN+L. This is a super annoying bug that rules out many potential combos involving the WIN key. 

I was thinking one potential solution would be to send an Esc input after the combination, but this also seems pretty jank.

ujl123, 2022-10-24:
> @ujl123 did you ever find a solution to this? I came across this same bug when trying to create a combo for WIN+L. This is a super annoying bug that rules out many potential combos involving the WIN [...]

I've never found a solution although a user in this thread had pointed out some solutions like flashing a dummy keycode with the override modifier, then sending over the new keycode might work, but that's beyond me.

If you do manage to make this work, please send a PR to have this fixed! I still like this feature even though I can't use GUI or ALT with it. Thanks!

wrinkledeth, 2022-10-25:
> The idea is good in that it will effectively prevent flashing modifiers and thus avoid opening the start menu or selecting the menus of a graphical app. I was just saying that we can't just hardcode [...]

@precondition Could you give me a pointer to how I could try implementing this fix? Would it be possible to solve this directly in the keymap without creating a new dummy variable (KEY_OVERRIDE_DUMMY)?

JonasGessner, 2022-12-11:
I ran across this same issue when trying to map mac-like keyboard shortcuts to windows keyboard shortcuts (i.e. GUI+C -> CTRL+C) and solved it like this:
https://github.com/JonasGessner/qmk_firmware/blob/jonas/users/jonas/jonas.h#L92
https://github.com/JonasGessner/qmk_firmware/blob/jonas/users/jonas/overrides.c#L58

Basically, if you customize the override using `custom_action` to send CTRL while still holding down GUI, and release GUI immediately after you prevent the start menu from showing while you release GUI. In your case where you don't want CTRL held down either I would assume you can release CTRL right thereafter and end up with no modifiers activated and no start menu shown.

precondition, 2023-05-31:
> Could you give me a pointer to how I could try implementing this fix? Would it be possible to solve this directly in the keymap?

@wrinkledeth Thankfully, key overrides provide a hook that you can use to work around the around as @JonasGessner showed in the previous comment on this thread but there are other features like Retro Tap which suffer from the same problem but do not provide any hook that one can use to solve the problem in user-space so I created #20992.

ujl123, 2023-07-09:
@precondition  Thanks for all the work on this! Does this mean that I can close this issue?

Also, looking at the PR it looks like this will require users to define in options dummy mods and what to neutralize. If I want to look it up in the wiki when it's finally in prod, will this be found in the retro tap features section, the key override section or in it's new dummy override section?

precondition, 2023-07-10:
The feature does not have its own separate page. It is documented in the tap-hold configurations page and the key overrides page.

Example: https://qmk.github.io/qmk_docs_devel/#/feature_key_overrides?id=neutralize-flashing-modifiers

ujl123, 2023-07-10:
Thank you! I had not realize that this was already a live change. I'll go ahead and make this as closed. Out of curiosity why was anti-flashing mods considered an opt in feature rather than it being baked into key overrides or retro tap? It seems like the expected behaviour of using LGUI + y = "a" is that LGUI doesn't flash. Is this because QMK has a principle of not breaking older configurations or is there is people who use key overrides that want to flash LGUI when they do LGUI +y ?

closed 2023-07-10 by ujl123

precondition, 2023-07-10:
It is a new feature that ever so slightly increases the firmware space and is not strictly necessary so it made more sense to set it as an opt-in feature. For example, the start menu of my desktop environment is brought up with <kbd>GUI</kbd>+<kbd>Space</kbd> so I don't care if <kbd>GUI</kbd> is being flashed. Extra code that would neutralize <kbd>GUI</kbd> flashes would simply be unnecessary bloat both in terms of firmware space and in terms of HID keyboard report packets sent to the host computer, in my case.

On another note, pay attention that this feature is live only in the `develop` branch right now. The documentation link I shared is representative of the state of the `develop` branch, not the main `master` branch. If you want to try out the feature now, you can check out the `develop` branch in your local copy of `qmk_firmware` with the terminal command `git checkout develop` and pull the latest changes with `git pull` (assuming your local `develop` branch is set to track the upstream remote `qmk/qmk_firmware` GitHub repository).

ujl123, 2023-07-10:
Thanks for the clarification on it still being in dev. I opened the link groggily on my phone and didn't notice the URL. 

On windows 10, any key override that uses LGUI is unusable as it will flash the start bar as well, but I had forgotten that in other OS and graphic interface environment flashing LGUI or alt does nothing, which makes your position make sense. I would argue that flashing mods is not a desired outcome even if the outcome nulll on some OSs  but as a space saving config it makes sense.