qmk/qmk_firmware #16799: [Bug] Keychron Q2 VIA keymap breaks keymap

Ladduro, 2022-04-05:
Flashing Keychron Q2 VIA keymap breaks the keymap. No key is working and VIA layout is empty even if loading the JSON provided by Keychron.

## Describe the Bug

Flashing Keychron Q2 VIA keymap breaks the keymap. No key is working and VIA layout is empty even if loading the JSON provided by Keychron. The same thing happens if flashing the default keymap with VIA_ENABLES set to yes. 

Board is recognized by VIA, but no keys are assigned.  Test matrix is working fine. Modifying RGB in VIA is working fine.

## System Information

**Keyboard:** Keychron Q2
**Revision (if applicable):** Rev 0110
**Operating system:** Win 11 / Fedora
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.

Ψ CLI version: 1.0.0

Ψ QMK home: /home/ionutmicu/Documents/GitHub/qmk_firmware

Ψ Detected Linux.

Ψ Git branch: q2-dev-personal

Ψ Repo version: 0.16.7

Ψ All dependencies are installed.

Ψ Found arm-none-eabi-gcc version 11.1.0

Ψ Found avr-gcc version 11.1.0

⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.

Ψ Found avrdude version 6.3

Ψ Found dfu-util version 0.11

Ψ Found dfu-programmer version 0.7.2

Ψ Submodules are up to date.

Ψ QMK is ready to go, but minor problems were found

```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

[image]

lokher, 2022-04-05:
Please refer to PR 15050: https://github.com/qmk/qmk_firmware/pull/15050, or use Keychron "playground" branch: https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/keychron

CrazyCoder, 2022-05-21:
The current playground branch linked above doesn't have persistent EEPROM. It does not freeze, but it also does not save any settings. Maybe @lalalademaxiya1 has any ideas how to fix it?

lalalademaxiya1, 2023-06-20:
@CrazyCoder Hi, since qmk supports the new EEPRO_DRIVER named wear_leveling now, which is more stable, we discarded the original algorithm and applied it to all Keychron Q series and V series keyboards.  You could find the newest source codes in Keychron "playground" branch. Thanks for your interest in Keychron and sorry for replying so late!

lalalademaxiya1, 2023-06-20:
@Ladduro Hi, now you could pull the latest source code from the Keychron "playground" branch or download the latest firmware from the Keychron official website for flashing. You won't encounter any similar issues anymore. Thank you for your assistance with Keychron, and we apologize for the delayed response. If you don't have any other questions, would you please close this PR at your convenience?

closed 2023-11-01 by zvecr