qmk/qmk_firmware #16821: [winry315] Bootloader not found.

aleksejrs, 2022-04-08:
In normal mode:
```
usbguard list-devices
40: block id f1f1:0315 serial "0" name "YD3xn15mx" hash "PL0UMEnIG3DHtnuDUWkU+Hz/V2Kt5RkdEHJ9xhJ6DU0=" parent-hash "WLLK8WnDcOZsmC13ldzWRmahoyPD7Y+TLn1uJ4TB2GU=" via-port "1-6" with-interface { 03:01:01 03:00:00 03:00:00 } with-connect-type "hotplug"
```

If plugged in with the top left button held:
```
usbguard list-devices
42: block id 03eb:2ff4 serial "1.0.0" name "ATm32U4DFU" hash "pRv+fV2QiheWoLTu53i8cbgowQBzc/ADFlkvxFk3AMI=" parent-hash "WLLK8WnDcOZsmC13ldzWRmahoyPD7Y+TLn1uJ4TB2GU=" via-port "1-6" with-interface 00:00:00 with-connect-type "hotplug"
```

In both modes:
```
Copying winry_winry315_default.hex to qmk_firmware folder                                           [OK]
Checking file size of winry_winry315_default.hex                                                    [OK]
 * The firmware size is fine - 26896/28672 (93%, 1776 bytes free)
Bootloader not found. Make sure the board is in bootloader mode. See https://docs.qmk.fm/#/newbs_flashing
 Trying again every 0.5s (Ctrl+C to cancel)...
```

The keyboard's buttons did nothing since the beginning. IIRC, when I booted the computer with it plugged in once, my mouse and graphics tablet's stylus didn't work until I unplugged it.

waffle87, 2022-04-08:
If you're on Linux, verify you have configured udev rules
[https://docs.qmk.fm/#/faq_build?id=cant-program-on-linux](https://docs.qmk.fm/#/faq_build?id=cant-program-on-linux)

aleksejrs, 2022-04-08:
I flashed it as root, but was still getting "interface 1 is not authorized for usage", and it didn't work until I edited  `/etc/usbauth.conf`, which was set to "deny HIDs if more than 2 are connected".

closed 2022-04-08 by aleksejrs