qmk/qmk_firmware #16829: [Feature Request] Reorganize the Sofle family folders

brianlow, 2022-04-09:
Reorganize the sofle code so keymaps are specific to hardware variation, all organized under a top level `sofle` folder

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Proposed folder structure:
```
keyboards/sofle/              # no code in this folder, just for grouping the family

    original/                 # moved from sofle/rev1                 
      keymaps/                      
        default/     
        flare56/              # no RGB code, so assume for the Sofle v1/v2
        helltm                # no RGB code, so assume for the Sofle v1/v2
        j4ckofalltrade        # no RGB code, so assume for the Sofle v1/v2
        killmaster            # no RGB code, so assume for the Sofle v1/v2
        manna-harbour_miryoku # no RGB code, so assume for the Sofle v1/v2

    rgb/                      # copy of sofle/rev1
      keymaps/
        default/              # renamed from rgb_default
        devdev/               # has RGB code, so assume for the Sofle RGB

    keyhive                   
      keymaps/
        default/              # copy of the current default keymap
        via/                  # renamed from keyhive_via

    choc                      # moved from from keyboards/sofle_choc
      keymaps
        default/
        via/   
```

Background
* When adding Sofle Choc support it was noticed that keymaps are organized under what I believe is an old folder structure (see [PR comment](https://github.com/qmk/qmk_firmware/pull/16736#discussion_r844724538)). The [recommended structure](
https://github.com/qmk/qmk_firmware/blob/master/docs/getting_started_introduction.md) is `/keymaps/family/hardware-variant/keymaps`

Pros
* keymaps are specific to the hardware variant
* hardware variants no longer share code so changes are isolated
* better CI since default/via keymaps will be built against hardware-variant specific code
* organized under a single family folder for easier discovery

Happy to make this change but want approval from hardware maintainers. If I don't heard back after a month I'll assume you are no longer maintaining and are ok with the change.

* Sofle v1/v2: @josefadamcik
* Sofle RGB: @DaneEvans
* Sofle Keyhive: @winder

winder, 2022-04-10:
This proposal looks good to me.

josefadamcik, 2022-04-11:
Looks great to me, thanks for doing this!

duese, 2022-08-15:
This issue is not stale. I created a PR (#17979) 5 days ago according to the proposal of @brianlow. I'm not sure how the process is now, who needs to review/approve that PR?

waffle87, 2022-08-16:
a collaborator who has some free time to review it. tldr; just need to be patient now.

closed 2022-12-16 by github-actions[bot]

ireun, 2023-04-25:
Any progress with that?

duese, 2023-04-25:
> Any progress with that?

I'm waiting for PR #16736 being accepted and merged. Than we can continue on restructuring the sofle family.