qmk/qmk_firmware #16832: [Bug] Rotary encoder on GMMK Pro always sends KC_VOL[U,D] in addtion to what's set by the keymap

mt-inside, 2022-04-11:
I have a GMMK Pro (ISO) with a "knob". If my `keymap.c` simulates tap of say `F12` for a wheel rotation I get a `VOLU` *plus* the `F12`. If the keymap simulates tapping `VOLU`, I get it twice. Same applies for counter-clockwise and `KC_VOLD`

Not sure if this a board-exclusive issue or a general problem with encoder support.

## Describe the Bug

## System Information

**Keyboard:** GMMK Pro (ISO)
**Revision (if applicable):** ?
**Operating system:** Linux
**`qmk doctor` output:** 
```
$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/matt/work/personal/src/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/matt/work/personal/src/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: matts-boards
Ψ Repo version: 0.16.8
⚠ Git has unstashed/uncommitted changes.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 11.2.0
Ψ Found avr-gcc version 11.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
no

## Additional Context
xev output for one clockwise click

With `tap_code()` lines commented out:
```
KeyPress event, serial 35, synthetic NO, window 0x4000001,
    root 0x1e9, subw 0x0, time 311449977, (1,597), root:(1729,637),
    state 0x0, keycode 123 (keysym 0x1008ff13, XF86AudioRaiseVolume), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 35, synthetic NO, window 0x4000001,
    root 0x1e9, subw 0x0, time 311449987, (1,597), root:(1729,637),
    state 0x0, keycode 123 (keysym 0x1008ff13, XF86AudioRaiseVolume), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

With `tap_code(KC_F12)`:
```
KeyPress event, serial 37, synthetic NO, window 0x4000001,
    root 0x1e9, subw 0x0, time 311590455, (-1,647), root:(1727,687),
    state 0x0, keycode 96 (keysym 0xffc9, F12), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

MappingNotify event, serial 38, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

KeyPress event, serial 38, synthetic NO, window 0x4000001,
    root 0x1e9, subw 0x0, time 311590465, (-1,647), root:(1727,687),
    state 0x0, keycode 123 (keysym 0x1008ff13, XF86AudioRaiseVolume), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

MappingNotify event, serial 38, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

KeyRelease event, serial 38, synthetic NO, window 0x4000001,
    root 0x1e9, subw 0x0, time 311590465, (-1,647), root:(1727,687),
    state 0x0, keycode 96 (keysym 0xffc9, F12), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

MappingNotify event, serial 40, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

KeyRelease event, serial 40, synthetic NO, window 0x4000001,
    root 0x1e9, subw 0x0, time 311590475, (-1,647), root:(1727,687),
    state 0x0, keycode 123 (keysym 0x1008ff13, XF86AudioRaiseVolume), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

fauxpark, 2022-04-11:
Should be fixed by #16830. If you've got your own keymap make the same changes to it as in that PR.

closed 2022-04-11 by fauxpark

mt-inside, 2022-04-11:
Oh yeah, annoyed I didn't spot that. Fixed, thanks