qmk/qmk_firmware #16840: [GMMK Pro] Setting "VIA_ENABLE = yes" in rules.mk breaks custom keycodes defined with #define statements

gourdo1, 2022-04-12:
## Describe the Bug

I borrowed some basic **#define** custom keycodes from the simple GMMK Pro ANSI keymap created by Ben Schaeff [here](https://github.com/qmk/qmk_firmware/tree/master/keyboards/gmmk/pro/ansi/keymaps/benschaeff), such as:

`#define SWAP_L SGUI(KC_LEFT) // Swap application to left display`

Despite copying them verbatim to my keymap.c file and asigning the aliases to my keymap matrix, I couldn't get them to work at all.

Through trial and error, I narrowed down the source of the problem to the 'VIA_ENABLE = yes' statement in my rules.mk. Sure enough, when I tried compiling Ben's simple keymap with the addition of 'VIA_ENABLE = yes' in /.../gmmk/pro/ansi/rules.mk, his #define statements also fail to engage. When i set it to 'no', all of our #define-based keycodes work as intended.

At this point, I'm reasonably confident it's either a bug or unsupported feature when VIA is enabled, but I don't know whether the scope of this incompatibility extends beyond the GMMK Pro, since that's the only keyboard I can test it on currently.

My temporary workaround is to disable VIA support in rules.mk, but I'd like to re-enable it if possible for greater customizability. There may be other ways to execute these custom keycodes, but I think #define statements should work with VIA enabled, as it's a pretty common way to build simple keycodes.

## System Information

**Keyboard:** GMMK Pro
**Revision (if applicable):** unknown
**Operating system:** Windows 10
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: C:/Users/me/Documents/GitHub/qmk_firmware
Ψ Detected Windows 10 (10.0.19044).
Ψ Git branch: PR-branch
Ψ Repo version: 0.16.8
⚠ Git has unstashed/uncommitted changes.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** No

fauxpark, 2022-04-12:
Did you try clearing the EEPROM?

On first run the VIA firmware copies the `keymaps` array in flash to the EEPROM, and switches to using that instead, as on AVR devices the flash memory cannot be rewritten at runtime. Therefore any changes to your keymap.c will not actually be reflected in the VIA app if there was already a valid keymap present in the EEPROM.

gourdo1, 2022-04-12:
> Did you try clearing the EEPROM?

I don't see a way to do this with QMK Toolbox. The "Clear EEPROM" button and menu are perpetually greyed out, even when I'm in bootloader mode, before and after I flash a new .bin file.

fauxpark, 2022-04-12:
Yes, STM32 devices have emulated EEPROM rather than a separate memory space like AVR microcontrollers, so `dfu-util` doesn't know how to write to it, and the Toolbox won't be able to either.
With VIA enabled Bootmagic Lite is also enabled, so you should be able to simply hold down the top left key while plugging the board in, which will clear the EEPROM and put the board into bootloader mode, then you can replug the board to boot normally.

Pop28, 2022-05-02:
> 

What do you mean by "simply hold down the top left key"? What key?

gourdo1, 2022-05-02:
@Pop28 Plug in while holding down the ESC key.

fauxpark, 2022-05-02:
I say "top left" because it is not always mapped to Escape. Generally it is the key at (0,0) in the electrical matrix.