qmk/qmk_firmware #16854: [Bug] KC_GRV or KC_GRAVE wrong key output on MacOS Monterey § and ±

akachida, 2022-04-14:
As the title said, I'm having trouble configuring the KC_GRV on MacOS Monterey with MBP M1 2020. Windows works fine.
Tried to include another layouts using language specific (I have a US Intl. keyboard) but no success.

The output that I got is § and ± (shifted)
I need a dead key like ', " and ^ that already works just fine.
Tried reseting EEPROM but didn't work and tried change my layout from OS too, still no success.
The only key with that problem is KC_GRV

## System Information

Keyboard: IDOBAO ID75 V2
Operating system: MacOS Monterey
`qmk doctor` output:
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /Users/angelochida/qmk_firmware
Ψ Detected macOS 12.3.1.
Ψ Git branch: master
Ψ Repo version: 0.16.9
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go

This is my layout
```bash
#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    [0] = LAYOUT_ortho_5x15(
        KC_ESC,  KC_GRV,   KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0,    KC_MINS, KC_EQL,  KC_DEL,
        KC_TAB,  KC_Q,     KC_W,    KC_F,    KC_P,    KC_G,    KC_COPY, KC_CUT,  KC_PSTE, KC_J,    KC_L,    KC_U,    KC_Y,    KC_SCLN, KC_BSPC,
        KC_CAPS, KC_A,     KC_R,    KC_S,    KC_T,    KC_D,    KC_VOLD, KC_MUTE, KC_VOLU, KC_H,    KC_N,    KC_E,    KC_I,    KC_O,    KC_QUOT,
        KC_LSFT, KC_Z,     KC_X,    KC_C,    KC_V,    KC_B,    KC_MPRV, KC_MPLY, KC_MNXT, KC_K,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_BSLS,
        KC_LGUI, KC_LCTL,  KC_LALT, KC_SPC,  KC_SPC,  KC_SPC,  KC_LBRC, MO(1),   KC_RBRC, KC_ENT,  KC_RALT, KC_LEFT, KC_UP,   KC_DOWN, KC_RGHT
    ),
    [1] = LAYOUT_ortho_5x15(
        KC_GRV, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, KC_F7, KC_F8, KC_F9, KC_F10, KC_F11, KC_F12, KC_PSCR, KC_NO,
        KC_NO,  KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO,  KC_NO,  KC_NO,  KC_NO,   KC_NO,
        KC_NO,  KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO,  KC_NO,  KC_NO,  KC_NO,   KC_NO,
        KC_NO,  KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO,  KC_NO,  KC_NO,  KC_NO,   KC_NO,
        KC_NO,  KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO,  KC_NO,  KC_NO,  KC_NO,   KC_NO
    ),
};
```

ujl123, 2022-04-14:
Do you have ISO set up? Double check to see if you have US English set up in your OS.

akachida, 2022-04-15:
Hello @ujl123, thank you for answering, and yes I tried ISO and ANSI with US English.
At first it show as grave, but when I press it, it changes do § on MacOS layout preview.

[image]

and after press it
[image]

ujl123, 2022-04-15:
It looks like the issue isn't with QMK but an on going issue with MacOS . Took a quick look and saw a bunch of stack overflow threads regarding it like : https://apple.stackexchange.com/questions/263882/how-to-fix-macos-sierra-language-switching-bug

There's also another thread that pressing FN toggles the layout but I'm unsure if that applies to you : https://apple.stackexchange.com/questions/416133/how-to-stop-fn-key-changing-my-keyboard-layout-macos-big-sur

Also why do you have U.S. and U.S. International --PC? Perhaps choosing  U.S. International --PC could fix it?

Sorry I couldn't help more, I'm not familiar with macs and how they deal with layouts. My only suspicion is that you might have an option to automatically detect and switch layout enabled somewhere, or perhaps you have some software that handles keymapping like karabiner that is messing up your layout.

fauxpark, 2022-04-15:
This is probably because you have that particular keyboard set in macOS as the wrong "type". There should be a button in keyboard preferences to switch it between ANSI, ISO and JIS, and you need to make sure you have the right keyboard "focused" by typing on it first. For example if you have two keyboards connected (or builtin & external keyboard) and open the input sources tab, if one is set as ISO and the other ANSI, you'll see the layout preview on the right change to match when you type on each. This is one of the quirks of how macOS handles keyboards - they each get their own configuration, and in fact shift and caps lock status are independent to each connected board.

akachida, 2022-04-15:
Well I tried everything that you guys said and still nothing seem to work.
The prints that I showed up there is like @fauxpark said, one is showing a ISO layout that is the built-in keyboard of my MBP, that is the default one, so is the first layout that is showing, and when I press the key from my idobao id75 it shows the ANSI layout that I configured from QMK.
What it's not clear is if it's a QMK problem or MacOS messing with key mapping.
I don't have any other software that change my key mapping.
Btw I have a bluetooth Keychron K2 that's working just fine.

fauxpark, 2022-04-15:
What you can do is press the "change keyboard type" button, then spam a key on the problem keyboard when it asks you to press a key, until it gives up and asks you to choose ANSI, ISO or JIS.

akachida, 2022-04-15:
Yes, already tried that, but even choosing ISO or ANSI or JIS, my macos always showing as ISO layout on keyboard preview

closed 2022-04-15 by akachida

fauxpark, 2022-04-15:
For future reference: http://m10lmac.blogspot.com/2009/12/fixing-keyboard-type-problems.html

blaisep, 2023-04-10:
I have the same issue on Monterey with my new v10 Alice....
I would not be quick to blame OS X because other keyboards work just fine.
[image]

I can force it to work if I run karabiner.... it's just one more layer of complexity.

> Yes, already tried that, but even choosing ISO or ANSI or JIS, my macos was always displaying as ISO layout on keyboard preview.
> 
> Edit: Oh man, well, there must be something wrong with the MacOS detecting layouts. So, basically I managed to work now. I tried to switch types in this order ANSI -> JIS(grave worked but messed with other keys) -> back to ISO and it detected as ISO again. Probably for some reason MacOS got lost on switching layouts.

blaisep, 2023-04-10:
FWIWI, @akachida , I would suggest describing this bug as
[image]

 "ANSI keyboard seen as ISO layout"
because the problem goes beyond those two keys.

Theorize, 2024-06-19:
I'm also having this issue.  I've tried changing the keyboard input to ABC, ABC-Extended, British English and US English. I've also tried changing the keyboard to be recognised as both ANSI and ISO.  But the grave key is still only recognised as §. Did you find a solution to this? I have another keyboard that uses QMK and the grave works correctly there.  I can't install Karabiner on my laptop.  Did anyone find a different solution to this?  My laptop language is set to UK-English, as this is what I use and I don't want to add US-English (since I don't use it).

fauxpark, 2024-06-19:
I already explained what you need to do above.