qmk/qmk_firmware #16855: ModTap or Tap Dance

pzrsa, 2022-04-15:
## Describe the Bug

## System Information

**Keyboard:** 
**Revision (if applicable):** 
**Operating system:** 
**`qmk doctor` output:** 
```
(Paste output here)
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

pzrsa, 2022-04-15:
accidental!

jtai, 2022-04-16:
I've experienced the same issue with my cantor. The problem can often (but not always) be triggered by putting my Mac to sleep for a few minutes. I think I may have also seen it once or twice during active use of the computer.

I was able to reproduce the issue with a variant of the very basic [default cantor keymap](https://github.com/qmk/qmk_firmware/pull/16552/files#diff-ca1ee26bbf3af9e00192da3b42f97f45b0ac016825385b83206c7ef0e85e1af6) that has one of the keys changed to a mod tap key. I have tried both the `master` branch with ChibiOS 20.3.4 and the `develop` branch with ChibiOS 21.11.1 but they both exhibit the issue.

I have also tried capturing debug logs by enabling console, enabling debug, and defining `DEBUG_ACTION`, but unfortunately I cannot seem to reproduce the problem when debugging is enabled.

pzrsa, 2022-04-16:
Thats exactly whats happening. Glad to hear I'm not the only one experiencing this issue.

For Tap Dance I can activate a single tap on the spacebar by pressing a modifier right before. So I tap CMD once and tap spacebar right after and it counts as a tap. Pressing again by itself or with a letter does not seem to do anything.

I believe this might a blackpill issue just with the other PR I linked as for that it was the result of the microcontroller.

Also on the mac login screen the tap works as it should, and then after logging in it goes back to not working.

jtai, 2022-04-16:
I am also relieved it's not just me since the issue seems somewhat obscure. 

Although my keymap has a tap dance key, I don't normally have a tap function assigned to it. But based on your comment, for experimentation purposes I assigned `KC_SPC` to it to see if it would also break, and it does. However, I also added some code to the tap dance handler to toggle the built in LED on the blackpill, and surprisingly, that part works. So it seems like the tap logic is fine -- it's just the key that doesn't get sent. That seems to line up with the findings in [this comment](https://github.com/qmk/qmk_firmware/issues/8259#issuecomment-595310253) on that older issue.

Here's the relevant code:

```
static bool dance_lower_led = false;

void dance_finished(qk_tap_dance_state_t *state, void *user_data) {
    switch (state->count) {
        case 1:
            if (!state->pressed) {
                dance_lower_led = !dance_lower_led;
                uint16_t level = dance_lower_led ? LED_PIN_ON_STATE : ~LED_PIN_ON_STATE;
                writePin(LED_CAPS_LOCK_PIN, level);

                tap_code(KC_SPC);
            }
            break;
```

You also need the following in `config.h`

```
#define LED_CAPS_LOCK_PIN C13
#define LED_PIN_ON_STATE 0
```

Finally, one difference in our experiences -- when it occurs, the issue affects me on the login screen after waking up from suspend and persists after logging in. In the password input if I tap the mod tap key, nothing happens.

rhruiz, 2022-04-21:
Same issue here. I am using a board with a STM32F411 mcu just like the blackpill that cantor uses

manna-harbour, 2022-09-25:
Seems to be the same issue as https://github.com/qmk/qmk_firmware/issues/16406.

jtai, 2025-03-24:
I have in my notes that I no longer experience the issue after updating to latest qmk in September 2023, but by that point I wasn't using the affected keyboard regularly. According to https://github.com/qmk/qmk_firmware/pull/20205#discussion_r1143972589, https://github.com/qmk/qmk_firmware/pull/19780 was the fix for the issue.