qmk/qmk_firmware #16861: [Feature Request] Adding a "Layer Lock" function. For when you momentary enter a layer but want to lock it in place.

ujl123, 2022-04-16:
This feature is for when you momentary enter a layer (ex: MO(1)) but you want to remain in the layer without having to hold down MO(1). Currently there is no clean solution to remain in the layer except to deactivate MO(1) and then use another keycode like TT(1) or TG(1).  I mention clean because there is in fact a solution of putting DF(1) on layer 1 to make it the default layer, but this might cause some unintentional effects rather than just locking layer 1 in place.

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x ] New behavior

## Description

https://getreuer.info/posts/keyboards/layer-lock/index.htm

Pascal Getreuer has already provided code snippets on how to do this, as well as an explanation of how it works, and why it's better than using DF(1).

To summarize functionality of the layer lock function:

1. Hold MO(1) to momentary enter layer 1 (But would work for LT(1,KC), OSL(1) or TT(1)
2. In layer 1, click on the keycode you've created for "Layer Lock". This will then keep you on layer 1
3. Do stuff on layer 1
4. Tap the "Layer Lock" key, or any other layer navigate key to break this lock.

getreuer, 2022-04-21:
Thanks for suggesting this! =)

Playing devil's advocate for a moment, the best reason I can come up with *against* adding Layer Lock to core is that existing features already give a few ways to get a similar effect. Besides `TT` and `DF` as you mention, one could use [Key Lock](https://docs.qmk.fm/#/feature_key_lock) to hold an `MO`.

On the other hand, the `DF` strategy has a risk of getting stuck in the layer if you forget to make an accessible switch to get back out. Key Lock has the caveat that "switching layers will not cancel the Key Lock." `TT` seems like the best existing solution. But for me at least, chording an `MO` key with a Layer Lock key is more comfortable than making multiple taps on a `TT` key, even if that's just two taps with `#define TAPPING_TOGGLE 2`. 

What are other folks' thoughts on this?

precondition, 2022-05-15:
Is there really a situation where you don't know in advance that you want to lock the layer? Besides, it doesn't cost you anything to accidentally hold down a `TT(layer)` key, realize your mistake and then release the key and then tap it in order to toggle the layer. what I've described takes the exact same amount of keystrokes as your suggestion.

closed 2022-10-20 by github-actions[bot]

jpsecher, 2023-01-24:
I would really love this Layer Lock feature.  

> I used to put a TO() or TG() key on the layers that I might want to lock in place and that does work just fine

I simply cannot get any of such alternatives to work.

As a minimal example, consider three layers; alpha, navigation and numpad: 

```
Layer 0
┌───────────────┬───────────────┐
│     KC_A      │     KC_B      │
├───────────────┼───────────────┤    
│ LT(1, KC_DEL) │ LT(2, KC_SPC) │
└───────────────┴───────────────┘

Layer 1
┌───────────────┬───────────────┐
│    KC_LEFT    │    KC_RGHT    │
├───────────────┼───────────────┤    
│    _______    │     TO(1)     │
└───────────────┴───────────────┘

Layer 2
┌───────────────┬───────────────┐
│     KC_0      │     KC_1      │
├───────────────┼───────────────┤    
│     TO(2)     │    _______    │
└───────────────┴───────────────┘
```

Holding the Delete key will switch to navigation, and holding the Space will switch to numpad, as expected.

But holding Delete and tapping Space will not lock the navigation layer.

I would really appreciate if anyone could tell me exactly how this can be achieved with existing features.

dcousens, 2024-09-05:
> Is there really a situation where you don't know in advance that you want to lock the layer?

@precondition if you are on a navigation layer (e.g layer 2), and you found your way there by using a home row modifier, how do you propose I can toggle that layer? I don't have any other keys available to allocate to `TO(2)` on my default layer.

If you hit `TO(2)` while on layer 2, `LT_2` key up will still revert the layer.  The only path I can see is setting `DF(2)`, but that will reset with my default layer settings each time; and I still don't have a path to `TO(0)` without using _another key_.

A layer lock key would help me prevent the momentary nature from leaving that layer, which is exactly what I want.  Then I can simply have a `TO(0)` to drop out of that layer when I'm ready.