qmk/qmk_firmware #16867: [Bug] After the WS2812 is powered on, the first light is always on green

biuboom-jun, 2022-04-17:
Describe the Bug
WS2812上电后第一个灯一直亮绿色，重新打开和关闭RGB效果后恢复正常
After the WS2812 is powered on, the first light is always on green, and it returns to normal after re-opening and closing the RGB effect

System Information
Keyboard:
Revision (if applicable):
Operating system:win10 20h2
Main control model: STM32F401RCT6
QMK Firmware version:0.13.34
Any keyboard related software installed?
 AutoHotKey
 Karabiner
[1 ] Other:VIA

closed 2022-10-20 by biuboom-jun

Timception, 2022-12-10:
I am also getting this using rgb matrix. hopefully we get an answer.

gezhaoyou, 2023-04-21:
同样遇到这个问题，解决了吗

biuboom-jun, 2023-04-21:
改用pwm控制就可以了

HorrorTroll, 2023-04-21:
If u use WS2812 at bitbang wiring, there a way bypass this issue by adding this code on <keyboard.c> file.
```
#ifdef RGB_MATRIX_ENABLE
bool process_record_kb(uint16_t keycode, keyrecord_t *record) {
        case RGB_TOG:
            if (record->event.pressed) {
                switch (rgb_matrix_get_flags()) {
                    case LED_FLAG_ALL: {
                        rgb_matrix_set_flags(LED_FLAG_NONE);
                        rgb_matrix_set_color_all(0, 0, 0);
                    }
                    break;
                    default: {
                        rgb_matrix_set_flags(LED_FLAG_ALL);
                        rgb_matrix_enable_noeeprom();
                    }
                    break;
                }
            }
            return false;
    }
    return process_record_user(keycode, record);
}

Another way is just use GPIO support PWM control.

gezhaoyou, 2023-04-21:
> 求教：pwd方式对芯片接RGB灯的引脚有具体要求吗？还是只要在 rules.mk 中指定 `WS2812_DRIVER = pwm`就可以

gezhaoyou, 2023-04-21:
求教：pwd方式对芯片接RGB灯的引脚有具体要求吗？还是只要在 rules.mk 中指定 WS2812_DRIVER = pwm就可以呢 @biuboom-jun

ganjianggaoshaoze, 2023-10-09:
我用的主控是apm32f072 ，pwm操纵rgb灯光但他还是一直亮绿灯啊