qmk/qmk_firmware #16894: I want to add new keyboard. but it is not available. Can you help me? Thank you

SEAN-JOU, 2022-04-20:
Pushing to https://github.com/qmk/qmk_firmware.git
remote: Permission to qmk/qmk_firmware.git denied to SEAN-JOU.
fatal: unable to access 'https://github.com/qmk/qmk_firmware.git/': The requested URL returned error: 403

zvecr, 2022-04-20:
You need to submit a `Pull Request`.

<https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request>
<https://opensource.com/article/19/7/create-pull-request-github>
<https://docs.qmk.fm/#/contributing?id=how-do-i-make-a-contribution>

closed 2022-04-20 by zvecr

SEAN-JOU, 2022-04-21:
I finished the pulling request.why I still haven’t successful