qmk/qmk_firmware #16899: [Bug] Compiling error in keymap.c

LoyalPotato, 2022-04-20:
When compiling, it spits out this error:
```
➜ qmk compile -kb loyalpotato/yggdrasil/rev1 -km LoyalPotato
Ψ Compiling keymap with make --jobs=1 loyalpotato/yggdrasil/rev1:LoyalPotato

QMK Firmware 0.16.8
Making loyalpotato/yggdrasil/rev1 with keymap LoyalPotato

avr-gcc (GCC) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/loyalpotato/yggdrasil/rev1/keymaps/LoyalPotato/keymap.c                       keyboards/loyalpotato/yggdrasil/rev1/keymaps/LoyalPotato/keymap.c:13:24: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'keymaps'
   13 | const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
      |                        ^~~~~~~
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:456: .build/obj_loyalpotato_yggdrasil_rev1_LoyalPotato/keyboards/loyalpotato/yggdrasil/rev1/keymaps/LoyalPotato/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:413: loyalpotato/yggdrasil/rev1:LoyalPotato] Error 1
```

## Describe the Bug

On compile get the error mentioned above and couldn't find any help as to what it could be. Only [this](https://github.com/qmk/qmk_firmware/issues/5024) and I've deleted the keyboard and run `qmk new-keyboard` again, but copied the files pertaining to the revision.
You can find the files [here](https://github.com/LoyalPotato/qmk_firmware).

## System Information

**Keyboard:** 
Yggdrasil (adding support)
**Revision (if applicable):** 
1
**Operating system:** 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/loyalpotato/Documents/keyboards/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.16.8
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 11.2.0
Ψ Found avr-gcc version 8.3.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go

```
## Additional Context

zvecr, 2022-04-20:
https://github.com/LoyalPotato/qmk_firmware/blob/master/keyboards/loyalpotato/yggdrasil/rev1/rev1.h is wrong.

Each `<keyboard>.h` will usually contain
```c
#pragma once

#include "quantum.h"
```

LoyalPotato, 2022-04-20:
@zvecr Yeah, based on seeing it everywhere I added and it cleared the error.
Can you tell me why I needed to add that?

zvecr, 2022-04-22:
Glad its now working!

`#pragma once` is to tell it to only process the file a single time.

`#include "quantum.h"` tells it to include a bunch of dependent definitions, one of which is for `PROGMEM` <https://www.nongnu.org/avr-libc/user-manual/pgmspace.html>.

closed 2022-04-22 by zvecr