qmk/qmk_firmware #16938: DZ60 possible layout question

chiboreache, 2022-04-25:
one button doesn't work, tested via `xev`

[image]

does it even support that way? or maybe it's just me and my soldering skills 

[image]
https://github.com/qmk/qmk_firmware/tree/master/keyboards/dz60

HorrorTroll, 2022-04-25:
It does support, but for some reason maker didn't make layout for that one.

chiboreache, 2022-04-25:
np, i create my own, but it's not working heh 😅

[image]

HorrorTroll, 2022-04-25:
U should modify dz60.h files too. Try replace this 
[image]

To this:
```
// 标准配列
/* Standard arrangement / LAYOUT
 * ┌───┬───┬───┬───┬───┬───┬───┬───┬───┬───┬───┬───┬───┬───┬───┐
 * │00 │01 │02 │03 │04 │05 │06 │07 │08 │09 │0a │0b │0c │0d │0e │
 * ├───┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴─┬─┴───┤
 * │10   │12 │13 │14 │15 │16 │17 │18 │19 │1a │1b │1c │1d │1e   │
 * ├─────┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴─────┤
 * │20    │22 │23 │24 │25 │26 │27 │28 │29 │2a │2b │2c │2d      │
 * ├──────┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬──┴┬───┬───┤
 * │30     │32 │33 │34 │35 │36 │37 │38 │39 │3a │3b │3c │3d │3e │
 * ├────┬──┴─┬─┴──┬┴───┴───┼───┴┬──┴───┴───┼───┼───┼───┼───┼───┤
 * │40  │41  │43  │44      │46  │48        │4a │4b │4c │4d │4e │
 * └────┴────┴────┴────────┴────┴──────────┴───┴───┴───┴───┴───┘
*/
#define LAYOUT( \
    k00, k01, k02, k03, k04, k05, k06, k07, k08, k09, k0a, k0b, k0c, k0d, k0e, \
    k10,      k12, k13, k14, k15, k16, k17, k18, k19, k1a, k1b, k1c, k1d, k1e, \
    k20,      k22, k23, k24, k25, k26, k27, k28, k29, k2a, k2b, k2c, k2d,      \
    k30,      k32, k33, k34, k35, k36, k37, k38, k39, k3a, k3b, k3c, k3d, k3e, \
    k40, k41,      k43, k44,      k46,      k48,      k4a, k4b, k4c, k4d, k4e  \
) { \
    { k00,  k01,   k02,   k03,  k04,  k05,   k06,  k07,   k08,  k09,   k0a,  k0b,  k0c,   k0d,  k0e   }, \
    { k10,  KC_NO, k12,   k13,  k14,  k15,   k16,  k17,   k18,  k19,   k1a,  k1b,  k1c,   k1d,  k1e   }, \
    { k20,  KC_NO, k22,   k23,  k24,  k25,   k26,  k27,   k28,  k29,   k2a,  k2b,  k2c,   k2d,  KC_NO }, \
    { k30,  KC_NO, k32,   k33,  k34,  k35,   k36,  k37,   k38,  k39,   k3a,  k3b,  k3c,   k3d,  k3e   }, \
    { k40,  k41,   KC_NO, k43,  k44,  KC_NO, k46,  KC_NO, k48,  KC_NO, k4a,  k4b,  k4c,   k4d,  k4e   }  \
}
```

chiboreache, 2022-04-25:
lol, got it!

chiboreache, 2022-04-25:
after replacing redundant `_______,` for actual `kc`  everything worked like a charm - thank you @HorrorTroll !!

i lived more than two years without one button.., what a shame..., but whatever =^-^=

chiboreache, 2022-04-25:
and now `qmk compile / qmk flash` also just works perfect! 

back then i have to use `sudo dfu-programmer atmega32u4 erase --force && sudo dfu-programmer atmega32u4 flash dz60_chiboreache.hex && sudo dfu-programmer atmega32u4 reset` 😬

HorrorTroll, 2022-04-25:
glad it work 👍

chiboreache, 2023-02-04:
@HorrorTroll 

plz update default template

```sh
➜  chiboreache git:(master) ✗ qmk compile -kb dz60 -km chiboreache
Ψ Compiling keymap with make --jobs=1 dz60:chiboreache

QMK Firmware 0.19.11
Making dz60 with keymap chiboreache

avr-gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/dz60/dz60.c                                                                    [OK]
Compiling: .build/obj_dz60/src/default_keyboard.c                                                   [OK]
Compiling: quantum/keymap_introspection.c                                                          In file included from ./.build/obj_dz60/src/default_keyboard.h:30,
                 from ./keyboards/dz60/keymaps/chiboreache/keymap.c:1,
                 from quantum/keymap_introspection.c:5:
./keyboards/dz60/keymaps/chiboreache/keymap.c:143:145: error: 'RESET' undeclared here (not in a function)
  143 |         _______        , KC_F1  , KC_F2  , KC_F3  , KC_F4  , KC_F5   , KC_F6  , KC_F7  , KC_F8  , KC_F9  , KC_F10  , KC_F11 , KC_F12 , _______, RESET,
      |                                                                                                                                                 ^~~~~
keyboards/dz60/dz60.h:28:97: note: in definition of macro 'LAYOUT'
   28 |     { k00,  k01,   k02,   k03,  k04,  k05,   k06,  k07,   k08,  k09,   k0a,  k0b,  k0c,   k0d,  k0e   }, \
      |                                                                                                 ^~~
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:360: .build/obj_dz60_chiboreache/quantum/keymap_introspection.o] Error 1
Make finished with errors
make: *** [Makefile:414: dz60:chiboreache] Error 1
```

HorrorTroll, 2023-02-04:
just change RESET to QK_BOOT

chiboreache, 2023-02-05:
@HorrorTroll 

```Compiling: quantum/unicode/unicode.c                                                               In file included from <command-line>:
./keyboards/dz60/keymaps/chiboreache/config.h:1:32: error: unsigned conversion from 'int' to 'unsigned char' changes value from '31795' to '51' [-Werror=overflow]
    1 | #define UNICODE_SELECTED_MODES UC_LINX
      |                                ^~~~~~~
quantum/unicode/unicode.c:69:34: note: in expansion of macro 'UNICODE_SELECTED_MODES'
   69 | static uint8_t selected[]     = {UNICODE_SELECTED_MODES};
      |                                  ^~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
```

so much new stuff `~_~`

chiboreache, 2023-02-05:
`#define UNICODE_SELECTED_MODES UNICODE_MODE_LINUX`

chiboreache, 2023-02-05:
haven't uploaded it yet, hope it works as expected😅

chiboreache, 2023-02-05:
nah, it shifts `z-row` again

and we are start from the beginning

moreover i can't turn off backlight (`red` for some reason), even with `BACKLIGHT_ENABLE = no` in options

chiboreache, 2023-02-06:
`LAYOUT`  -> `LAYOUT_directional` do the job👌

[image]

[image]

closed 2023-02-06 by chiboreache