qmk/qmk_firmware #16940: SEND_STRING send nothing [Bug]

chiboreache, 2022-04-25:
## Describe the Bug

everything is compiles and flashes just fine, but after pressing the button - some OS things happen (muting sound and volume up (f13 f14)) but with shift (regular kc in upper layer) it's ok

## System Information

**Keyboard:** 
dz60

**Operating system:** 
Arch Linux Linux 5.17.4-arch1-1
KDE 5.93.0 / Plasma 5.24.4

**`qmk doctor` output:** 
```
➜  ~ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/chibo/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.16.9
⚠ Git has unstashed/uncommitted changes.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 11.3.0
Ψ Found avr-gcc version 11.3.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
➜  ~
```

**Any keyboard related software installed?** 
no

## Additional Context

yeah code is tough, that was hard work, mostly of [ridingqwerty](https://github.com/ridingqwerty) (https://github.com/qmk/qmk_firmware/issues/7839)

i see some progress, so i revisit it later, but if you help me organize it more simple way (like `[declaration, [lower, upper]]`  w/o that low-level parametric stuff `tap_code16(alt_keymap[1][keycode - AKM_1_8]);` - i would be much appreciated   

```c

enum {
  TD_SHIFT_CAPS,
};

enum custom_keycodes {

  ARROW_FUNC = SAFE_RANGE,
  ARROW_RETURN,

//   LANG,
  RUS_LAT,
// };
// enum shifted_keycodes {
  AKM_1_8,

  AKM_1_9,
//   AKM_1_10,
//   AKM_1_11,
//   AKM_1_12,
  K_F13,
  K_F14,

  AKM_2_22,
  AKM_2_23,

  AKM_3_11,

  AKM_4_11,

  M_CYR_ER,

};

// Tap Dance definitions
qk_tap_dance_action_t tap_dance_actions[] = {
    // Tap once for Escape, twice for Caps Lock
    [TD_SHIFT_CAPS] = ACTION_TAP_DANCE_DOUBLE(KC_LSHIFT, KC_CAPS)
};

// const int len_enum = end - start + 1;
// uint16_t alt_keymap[2][len_enum] = {
uint16_t alt_keymap[2][10] = {
        {
            // AKM_1_8
            KC_ASTERISK,

            // AKM_1_9
            KC_LEFT_PAREN,

            // KC_UNDERSCORE,
            // KC_MINUS,
            // KC_EQUAL,

            // K_F13
            ARROW_FUNC,

            // K_F14
            ARROW_RETURN,

            KC_LBRACKET,
            KC_LEFT_CURLY_BRACE,

            KC_COLON,

            KC_SLASH,

            // CYR
            KC_M,
        },{
            // AKM_1_8
            KC_AMPERSAND,

            // AKM_1_9
            KC_RIGHT_PAREN,

            // KC_UNDERSCORE,
            // KC_MINUS,
            // KC_EQUAL,

            // K_F13
            KC_BSLASH,

            // K_F14
            KC_PIPE,

            KC_RBRACKET,
            KC_RIGHT_CURLY_BRACE,

            KC_SCOLON,

            KC_PIPE,

            // CYR
            KC_RBRACKET,
        }
};

uint16_t key_timer;
bool process_record_user(uint16_t keycode, keyrecord_t *record)
{
  switch (keycode)
  {
    case RUS_LAT:
        if (record->event.pressed)
            {
                key_timer = timer_read();
                register_mods(MOD_LSFT);
            }
        else
            {
                unregister_mods(MOD_LSFT);
                if (timer_elapsed(key_timer) < 500)
                    {
                    layer_invert(RUS);
                    tap_code16(C(G(KC_NO)));
                    }
                else
                    {
                    }
            }
        return false;
        break;

    uint8_t mod_state = get_mods();
    case AKM_1_8 ... M_CYR_ER:
      if (record->event.pressed)
      {
        mod_state = get_mods() & MOD_MASK_SHIFT;
        clear_mods();
        if(mod_state)
        {
          tap_code16(alt_keymap[1][keycode - AKM_1_8]);
        }
        else
        {
          tap_code16(alt_keymap[0][keycode - AKM_1_8]);
        }
        set_mods(mod_state);
      }
      return false;

    case ARROW_FUNC:
        if (record->event.pressed) {
        SEND_STRING("=>");
        }
        break;

    case ARROW_RETURN:
        if (record->event.pressed) {
        SEND_STRING("->");
        }
        break;
  }
  return true;
};

```

chiboreache, 2022-04-26:
i found !

solo macros works just fine ツ := = => ->

but in that `lower/upper workaround` it's simply going BEFORE registration, that's why in macros only stock KC are working..

____

please halp me with dataflow !

precondition, 2022-05-15:
You can't pass custom keycodes like `M_ARROW_RETURN` to `tap_code16`