qmk/qmk_firmware #16986: [Bug] Teensy LC Ez-maker Direct Pin will not flash (HEX PARSE ERROR LINE 3)

Xecuted-Toast, 2022-05-02:
I cannot get QMK Configurator to make a hex file (it creates a .bin). I am unable to open and review the .bin file. 
When trying to flash the Teensy LC with the .bin file i get from compile. I get this error. "HEX PARSE ERROR LINE 3"

[Screen Shot 2022-05-01 at 9.30.06 PM.pdf](https://github.com/qmk/qmk_firmware/files/8600526/Screen.Shot.2022-05-01.at.9.30.06.PM.pdf)

Any help is greatly appreciated!

## System Information

handwired TarTarus pro RAzer with choc switches
[IMG_1106.pdf](https://github.com/qmk/qmk_firmware/files/8600539/IMG_1106.pdf)

fauxpark, 2022-05-02:
This would be partially fixed by https://github.com/qmk/qmk_firmware/pull/14420 - it sets `FIRMWARE_FORMAT = hex` for all of the ARM Teensys. However you will still not be able to flash it in the Toolbox at it only has MCU values for AVR parts.

For now, you will have to manually add the above line to your rules.mk, and use the Teensy Loader GUI to flash.

Xecuted-Toast, 2022-05-02:
When I open teensyduino I am unable to bring the .bin file into it. Is the teenysduino the Teensy Loader GUI? When attempting with the traditional teensy loader the program key is greyed out when the .bin file is selected. Is there a video tutorial that could assist me? I am feeling like a bit of a noob with the software end of things. I still feel a bit lost?

fauxpark, 2022-05-02:
No, you need a .hex file.

Xecuted-Toast, 2022-05-02:
so I added the line 108 to 111 on the builddefs/bootloader.mk to make the LC teensy changes 
"# Teensy LC, 3.0, 3.1/2, 3.5, 3.6
    ifneq (,$(filter $(MCU_ORIG), MKL26Z64 MK20DX128 MK20DX256 MK64FX512 MK66FX1M0))
        FIRMWARE_FORMAT = hex
    endif"
but how do I get a hex file, I have no way to alter the way QMK configurator runs and makes a .bin; sorry I am still a bit lost?

fauxpark, 2022-05-02:
You won't be able to do this in the Configurator; you need to compile it yourself.

Xecuted-Toast, 2022-05-04:
when picking a layout to compile for .hex with MSYS2 can I use a .Json made with configurator? And I do not attempt to flash with qmk toolbox, but need to try teensy flasher or qmk flash directly with MSYS2?

Xecuted-Toast, 2022-05-05:
[image]
So now I am getting : <qmk.exe compile: error: argument -kb/--keyboard: invalid keyboard_folder value: 'ez_maker/directpins/teensy_lc'>

I am not sure why this is happening in MSYS264bit?

fauxpark, 2022-05-05:
Does that folder exist?

Xecuted-Toast, 2022-05-05:
yes qmk_firmware/keyboards/ez_maker/directpins/teensy_lc

fauxpark, 2022-05-05:
Have you modified it in some way? What's the output of `git status` from your `qmk_firmware` folder?

Xecuted-Toast, 2022-05-05:
do I need to set up git to get it to work? I have not forked yet

Xecuted-Toast, 2022-05-05:
no modifications yet

fauxpark, 2022-05-05:
Did you download the zip file from Github then? That's not going to work. You need to follow the newbs guide in the docs.

Xecuted-Toast, 2022-05-05:
I downloaded zip from github and changed rules.mk like we talked about for .hex within the teensy_lc folder. I went through newbs section

fauxpark, 2022-05-05:
I did not tell you to download the zip.

Xecuted-Toast, 2022-05-05:
Yes I am pretty terrible at this and following directions.  I deleted everything, forked with git. Then updated folder and still get the same qmk.exe compile error?

Xecuted-Toast, 2022-05-05:
[image]

Xecuted-Toast, 2022-05-05:
[image]

fauxpark, 2022-05-05:
Please post the output of `qmk doctor`.

Xecuted-Toast, 2022-05-05:
[image]
Thank you for all your help! I am learning but still having trouble when I hit a bump

fauxpark, 2022-05-06:
So that's claiming there are modifications, what is the output of `git status` as well?

Xecuted-Toast, 2022-05-06:
[image]

 //So I did add the line "FIRMWARE_FORMAT = hex "
//"rules.mk":
# MCU name
USE_CHIBIOS_CONTRIB = yes
FIRMWARE_FORMAT = hex 
# Enter lower-power sleep mode when on the ChibiOS idle thread
OPT_DEFS += -DCORTEX_ENABLE_WFI_IDLE=TRUE

Xecuted-Toast, 2022-05-06:
it appears to have compiled hex! will give update tomorrow :)

Xecuted-Toast, 2022-05-07:
IT WORKS! You helped me sooo much! Thank you !!!!!!

closed 2022-05-07 by Xecuted-Toast