qmk/qmk_firmware #16995: [Bug] qmk setup fails on Apple M1 (Pro), bad CPU type `arm-none-eabi-gcc`

Siilwyn, 2022-05-03:
## Describe the Bug
After a successful `brew install qmk/qmk/qmk` running `qmk setup` fails with the following output:
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /Users/siilwyn/qmk_firmware
Ψ Detected macOS 12.3.1.
Ψ Git branch: master
Ψ Repo version: 0.16.9
Ψ CLI installed in virtualenv.
<class 'OSError'>
☒ [Errno 86] Bad CPU type in executable: 'arm-none-eabi-gcc'
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/qmk/1.0.0_1/libexec/lib/python3.9/site-packages/milc/milc.py", line 514, in __call__
    return self.__call__()
  File "/opt/homebrew/Cellar/qmk/1.0.0_1/libexec/lib/python3.9/site-packages/milc/milc.py", line 519, in __call__
    return self._subcommand(self)
  File "/Users/siilwyn/qmk_firmware/lib/python/qmk/cli/doctor/main.py", line 98, in doctor
    bin_ok = check_binaries()
  File "/Users/siilwyn/qmk_firmware/lib/python/qmk/cli/doctor/check.py", line 102, in check_binaries
    if not is_executable(binary):
  File "/Users/siilwyn/qmk_firmware/lib/python/qmk/cli/doctor/check.py", line 142, in is_executable
    check = cli.run([command, version_arg], combined_output=True, stdin=DEVNULL, timeout=5)
  File "/opt/homebrew/Cellar/qmk/1.0.0_1/libexec/lib/python3.9/site-packages/milc/milc.py", line 161, in run
    return subprocess.run(command, **kwargs)
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 86] Bad CPU type in executable: 'arm-none-eabi-gcc'
```

## System Information

**Operating system:** MacOS Monterey
**`qmk doctor` output:** same as output above.

`which avr-gcc && file "$(which avr-gcc)"` returns:
```
/opt/homebrew/bin/avr-gcc
/opt/homebrew/bin/avr-gcc: Mach-O 64-bit executable arm64
```

## Additional Context

Trying to install qmk on a fresh MacOS install.

fauxpark, 2022-05-03:
You need to install Homebrew through Rosetta.

Siilwyn, 2022-05-03:
Ah that's unfortunate, I'm trying to avoid Rosetta so far. Is this an issue that won't be fixed because of the unsupported CPU type? (As in, should I close the issue?)

fauxpark, 2022-05-03:
See https://github.com/osx-cross/homebrew-avr/issues/232

closed 2022-05-03 by Siilwyn

reinier, 2022-12-02:
Any new options on this? Trying to install QMK on a fresh MacBook Apple Silicon and trying to avoid Rosetta.

fauxpark, 2022-12-02:
No, GitHub Actions still has no M1 runners so bottles cannot be built for it. Also, the `qmk` formula still specifies `arm-gcc-bin@8` which is a precompiled toolchain directly from the ARM site.