qmk/qmk_firmware #17002: [Feature Request] Support for Keychron K4 V2

Bianks84, 2022-05-04:
As the title states, looking for support for Keychron K4.
The keyboard supports RGB and hot-swaps if that helps at all.

New to GitHub and the hobby so let me know if anything else is needed!
Thanks in advance!

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x ] New behavior

## Description

MikeWrock, 2022-07-08:
I have a fork that works on my k4v2 RGB, it's a hacked together mix of the sonix_qmk fork and this one to get CAPSWORD functionality. Tested on Ubuntu 22.04, I had to fork the sonix_flasher repo as well most likely due to my version of Ubuntu and python 2's deprecation. Give it a shot and let me know if it works

MikeWrock, 2022-07-11:
> Appreciate the reply, like I mentioned very new to this so once I have a
> clue of what you just said I'll get back to you.
> Always willing to test it on an Ubuntu VM before seeing if it'll work on
> Windows.
> Going through the repos you mentioned will hopefully give me more of a
> clue.
> Thanks!

The steps I used I wrote down in the readme

closed 2022-08-29 by tzarc