qmk/qmk_firmware #17045: [Bug] one shot modifier isn't timed out after clicking with mouse

rhhub, 2022-05-09:
## Describe the Bug
[config and keymap](https://github.com/rhhub/qmk_firmware/tree/master/keyboards/ferris/keymaps/rhhub)

1. tap OSM(MOD_LCTRL)
2. click link in mozilla to open link in new tab
3. scroll wheel will change zoom level (expected to scroll page) until i consume osm by clicking a letter; it won't time out.

## System Information

**Keyboard:ferris sweep** 
**Revision (if applicable):** 
**Operating system:windows 10** 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: C:/Users/ryanh/_home/qmk_firmware
Ψ Detected Windows 10 (10.0.19044).
Ψ Git branch: master
Ψ Repo version: 0.16.9
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [x] AutoHotKey (Windows) not running
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

drashna, 2022-10-07:
Could you see if this is still happening on develop?

rhhub, 2022-10-07:
I can't I went wireless with zmk.

KoFish, 2024-02-29:
Since the setup does not seem to even include a mouse config within QMK I assume this is about using a separate mouse. In that case there is no way for the OSM to be consumed since the mouse click does not send a signal to the keyboard. As far as the computer cares the modifier is just held down, release is done by internal mechanisms within the keyboards firmware.