qmk/qmk_firmware #17048: [Feature Request]  Support for Keychron K3 V2

SamuelChih, 2022-05-10:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

As the title states, I would like to see support for the Keychron K3 V2. This keyboard have the screenshot key locate directly above backspace. It is very easy to accidentally hit it, and stuck in edit screenshot screen on iOS device and there is no way to remap/disable the key in iOS.

HorrorTroll, 2022-05-10:
Dunno how to say, but there are unofficial porting QMK for those keeb by community anyway. But it when u use QMK, u will lost the way to use BLE mode.

https://github.com/SonixQMK/qmk_firmware/tree/sn32_master/keyboards/keychron/k3

amrsoll, 2022-06-16:
They already have used QMK for some of their keyboards. You should let them know they need to support QMK for yours too :)

tzarc, 2022-06-16:
Best bet for adding support for these is to do it yourself and then submit back to QMK.
There are hundreds of other outstanding items at this point and none of QMK's team are likely to pick up something like this.

As suggested, the SonixQMK fork may well be your answer -- you'd be better of looking at their side of things at this point.

nahuel, 2022-08-18:
Check this:  https://seyahdoo.com/blog/qmk-on-keychron-k3v2/

closed 2022-08-29 by tzarc