qmk/qmk_firmware #17051: error: unknown type name 'macro_t'

arialdomartini, 2022-05-10:
I'm getting compilation errors with a freshly cloned qmk_firmware repository, on all the keymaps using macro.

I started from a fresh clone of this repo, then I run

```bash
sudo ./util/docker_build.sh dz60:n0velty
```

This results in

```
QMK Firmware 0.16.9
Making dz60 with keymap n0velty

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/dz60/dz60.c                                                                    [OK]
Compiling: keyboards/dz60/keymaps/n0velty/keymap.c                                                 In file included from keyboards/dz60/keymaps/n0velty/keymap.c:1:0:
keyboards/dz60/keymaps/n0velty/keymap.c:38:116: error: implicit declaration of function 'M' [-Werror=implicit-function-declaration]
        _______,          _______,                   _______,                                     _______,          M(0)),
                                                                                                                    ^
keyboards/dz60/dz60.h:114:81: note: in definition of macro 'LAYOUT_true_hhkb'
     { KC_NO, k41,   KC_NO, k43,  KC_NO, KC_NO, k46,  KC_NO, KC_NO, KC_NO, k4a,  k4b,  KC_NO, KC_NO, KC_NO }  \
                                                                                 ^
keyboards/dz60/keymaps/n0velty/keymap.c:38:116: error: initializer element is not constant
        _______,          _______,                   _______,                                     _______,          M(0)),
                                                                                                                    ^
keyboards/dz60/dz60.h:114:81: note: in definition of macro 'LAYOUT_true_hhkb'
     { KC_NO, k41,   KC_NO, k43,  KC_NO, KC_NO, k46,  KC_NO, KC_NO, KC_NO, k4a,  k4b,  KC_NO, KC_NO, KC_NO }  \
                                                                                 ^
keyboards/dz60/keymaps/n0velty/keymap.c:38:116: note: (near initialization for 'keymaps[2][4][11]')
        _______,          _______,                   _______,                                     _______,          M(0)),
                                                                                                                    ^
keyboards/dz60/dz60.h:114:81: note: in definition of macro 'LAYOUT_true_hhkb'
     { KC_NO, k41,   KC_NO, k43,  KC_NO, KC_NO, k46,  KC_NO, KC_NO, KC_NO, k4a,  k4b,  KC_NO, KC_NO, KC_NO }  \
                                                                                 ^
keyboards/dz60/keymaps/n0velty/keymap.c:42:7: error: unknown type name 'macro_t'
 const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) {
       ^
keyboards/dz60/keymaps/n0velty/keymap.c: In function 'action_get_macro':
keyboards/dz60/keymaps/n0velty/keymap.c:49:12: error: 'MACRO_NONE' undeclared (first use in this function)
     return MACRO_NONE;
            ^
keyboards/dz60/keymaps/n0velty/keymap.c:49:12: note: each undeclared identifier is reported only once for each function it appears in
keyboards/dz60/keymaps/n0velty/keymap.c:49:12: error: return from incompatible pointer type [-Werror=incompatible-pointer-types]
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:456: .build/obj_dz60_n0velty/keyboards/dz60/keymaps/n0velty/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:413: dz60:n0velty] Error 1
```

Any idea how I could proceed?

zvecr, 2022-05-10:
<https://docs.qmk.fm/#/ChangeLog/20220226?id=legacy-macro-and-action_function-system-removed-16025>

arialdomartini, 2022-05-10:
Thank you @zvecr! 
Now I understand why the macros in that keymap did not match the documentation.

I wrongly assumed that all the keymaps in master could be successfully built.

closed 2022-05-10 by arialdomartini