qmk/qmk_firmware #17059: The keyboard name cannot be set to F12, the compilation can pass normally, but the computer recognizes it as a string of messy letters

jeromeli002, 2022-05-11:
The keyboard name cannot be set to F12, the compilation can pass normally, but the computer recognizes it as a string of messy letters
键盘名字不能设置F12，编译能正常通过，但是电脑识别出来是一串乱七八糟的字母

fauxpark, 2022-05-11:
This is a known issue, it has to do with how the USB manufacturer and product strings are defined in the config, as bare text:
```c
#define MANUFACTURER Keyboard Maker
#define PRODUCT Keyboard Name
```
rather than string literals:
```c
#define MANUFACTURER "Keyboard Maker"
#define PRODUCT "Keyboard Name"
```
Which means the preprocessor will try to evaluate it and potentially mangle the result.
Fixing this is on my todo list but it involves touching pretty much every keyboard in the repo at the same time as changing the core code that deals with these defines, so it may take some time.

fauxpark, 2022-09-26:
This has been fixed in the develop branch.

closed 2022-09-26 by fauxpark