qmk/qmk_firmware #17064: platforms/avr/gpio.h:33:46: error: array subscript 0 is outside array bounds

arialdomartini, 2022-05-11:
I'm not sure if the following is an issue with the Arch `qmk` package, or whatever.

The following successfully compiles:

```bash
sudo ./util/docker_build.sh dz60:default
```

Instead,

```bash
qmk compile -c -kb dz60 -km default
```
results in

```
Compiling: quantum/led.c                                                                           In file included from platforms/gpio.h:21,
                 from quantum/led.c:19:
quantum/led.c: In function 'led_update_kb':
platforms/avr/gpio.h:33:46: error: array subscript 0 is outside array bounds of 'volatile uint8_t[0]' {aka 'volatile unsigned char[]'} [-Werror=array-bounds]
   33 | #define writePinLow(pin) (PORTx_ADDRESS(pin) &= ~_BV((pin)&0xF))
      |                                              ^~
platforms/avr/gpio.h:34:61: note: in expansion of macro 'writePinLow'
   34 | #define writePin(pin, level) ((level) ? writePinHigh(pin) : writePinLow(pin))
      |                                                             ^~~~~~~~~~~
quantum/led.c:96:9: note: in expansion of macro 'writePin'
   96 |         writePin(LED_CAPS_LOCK_PIN, led_state.caps_lock);
      |         ^~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:456: .build/obj_dz60_default/quantum/led.o] Error 1
Make finished with errors
make: *** [Makefile:413: dz60:default] Error 1
```

Any ideas?
Is this an issue I should open to the Arch qmk package maintainer?

rti, 2022-05-11:
Which avr-gcc version are you running? I encountered a similar problem with avr-gcc 12. My workaround was to downgrade to avr-gcc 11.

arialdomartini, 2022-05-11:
```
avr-gcc (GCC) 12.1.0
```

I'm on a rolling distro. I believe that downgrading won't be that easy.

7h3w1zz, 2022-05-12:
I just had the same issue.

> I believe that downgrading won't be that easy.

As a temporary workaround, downgrade should be `pacman -U /var/cache/pacman/pkg/avr-gcc-11.3.0-1-x86_64.pkg.tar.zst` (Assuming you keep at least one previous package version around).  I tested rebuilding and flashing my keyboard's firmware, and everything seems to be working...

rti, 2022-05-12:
You can also exclude the package from further upgrades in `/etc/pacman.conf`
```
 IgnorePkg = avr-gcc
```

Brainzman, 2022-05-12:
> I just had the same issue.
>
>
> As a temporary workaround, downgrade should be `pacman -U /var/cache/pacman/pkg/avr-gcc-11.3.0-1-x86_64.pkg.tar.zst` (Assuming you keep at least one previous package [...]

If you do not have `avr-gcc` in your cache, you can use the [Archlinux Archive](https://wiki.archlinux.org/title/Arch_Linux_Archive) and downgrade using:

`pacman -U https://archive.archlinux.org/packages/a/avr-gcc/avr-gcc-11.3.0-1-x86_64.pkg.tar.zst`

Please note that this problem *seems* to come from the fact that GCC introduced a new warning, or added more checks to their `array-bounds` warning. And the `-Wall -Werror` qmk triggers it.

As such, a proper solution would be a patch that either fixes `i2c_master.c` and others, or disable the warning. I'll try to see if I can throw something together tomorow.

> You can also exclude the package from further upgrades in `/etc/pacman.conf`
> 
> ```
>  IgnorePkg = avr-gcc
> ```
I would advise against this fix, unless you run `pacman -Syu` everyday, or you can remember that you have the package ignored.

zvecr, 2022-05-14:
In the short term, you can also disable the error locally:

```bash
make clean dz60:default AVR_CFLAGS="-Wno-array-bounds"
```
or...
```
qmk compile -c -kb dz60 -km default -e AVR_CFLAGS="-Wno-array-bounds"
```

sigprof, 2022-05-14:
Apparently this is https://gcc.gnu.org/bugzilla/show_bug.cgi?id=105523

The workaround suggested there is `--param=min-pagesize=0`; unfortunately, while this option apparently fixes compilation with avr-gcc 12.1, it also breaks all older versions of avr-gcc, because the option was added only in 12.x and 11.3 (see https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578). So if we choose to add this option, we would need to add some `$(shell $(CC) ...)` tricks to check whether that option is actually supported by the used compiler.

mcp292, 2022-05-26:
Same issue on Arch, avr version 12.1, compiling for Fortitude60:
```txt
Compiling: platforms/avr/drivers/i2c_master.c                                                      In function 'i2c_start_impl',
    inlined from 'i2c_start' at platforms/avr/drivers/i2c_master.c:101:18:
platforms/avr/drivers/i2c_master.c:58:10: error: array subscript 0 is outside array bounds of 'volatile uint8_t[0]' {aka 'volatile unsigned char[]'} [-Werror=array-bounds]
   58 |     TWCR = 0;
```

zvecr, 2022-05-29:
Closing as https://github.com/qmk/qmk_firmware/pull/17136 is now on master.

closed 2022-05-29 by zvecr